projectacrn/acrn-hypervisor #1310: ACRN Service OS not booting on Intel NUC6CAYH

JeremyHigginsAllion, 2018-09-20:
I am attempting to get the ACRN Hypervisor to function on an Intel NUC6CAYH.

I have successfully followed the instructions found [here](https://projectacrn.github.io/latest/getting-started/apl-nuc.html) up to Step 7.

After the Step 7 command `# systemctl start weston@cl_sos` is entered, a Desktop is displayed. Step 8 instructs that I reboot the device at this point, then select `The ACRN Service OS`. When I do so, the screen goes black for exactly 60 seconds, then returns to the OS selection screen.

If I select the second option, `Clear Linux OS for Intel Architecture (Clear-linux-native-4.17.6.590)` , I get `[     0.274145] kvm:  no hardware support`.

If I select the third option, `Clear Linux OS for Intel Architecture (Clear-linux-pk414-sos-4.14.52.63)`, I get a brief flash of text, too quick to read, followed by a black screen.

If I select the fourth option, `Clear Linux OS for Intel Architecture (Clear-linux-pk414-standard-4.14.52.63)`, I get the same Desktop that Step 7 produced before the reboot.

Further, the fifth option, `EFI Default Loader` (shown in the instructions), does not appear on my list.

A colleague of mine attempted to continue the instructions from Step 8 onward, while using the command line available from the Desktop produced by Step 7, but the process then failed on "Set up Reference UOS", Step 5.

I am using the ClearLinux Build 25090, found [here](https://download.clearlinux.org/image/), specifically [this](https://download.clearlinux.org/image/clear-25090-installer.img.xz) image.

Is there any way I can collect a print out of the NUC boot logs and/or the ACRN Service OS boot logs?

mattcurfman, 2018-10-02:
Clear Linux version numbers:
Build 25180
sos.4.14.69-102
standard.4.14.69-102

NUC6CAYH
Total Memory: 4GB
BIOS Version: AYAPLCEL.86a.0055.2018.0821.1152
Processor: Intel Celeron CPU J3455 @ 1.50 GHz

Perhaps related, I'm noticing that the SOS kernel takes over a minute to boot, and I originally thought it was hung like @JeremyHigginsAllion noted above.  Attached below is a portion of the syslog.  The same behavior was **not** noted on the UP2 board.  Only known difference is that UART set to 'disabled' in EFI boot mgr since this system doesn't have a serial port.

```
[    0.555360] idma64 idma64.0: Found Intel integrated DMA 64-bit
[    0.560632] idma64 idma64.1: Found Intel integrated DMA 64-bit
[    0.561331] console [ttyS0] disabled
[    0.561635] dw-apb-uart.1: ttyS0 at MMIO 0x91420000 (irq = 4, base_baud = 115200) is a 16550A
[    0.851377] random: fast init done
[   47.968958] console [ttyS0] enabled
[   48.016426] idma64 idma64.2: Found Intel integrated DMA 64-bit
[   48.095543] idma64 idma64.3: Found Intel integrated DMA 64-bit
[   48.174552] idma64 idma64.4: Found Intel integrated DMA 64-bit
[   48.253167] hyper_dmabuf_starting: Initialization started
```

gvancuts, 2018-10-17:
Hi @JeremyHigginsAllion , sorry for the delay in getting back to you. I did forget to follow up with you on this.

The problem whereby the system goes back to the OS selection menu could be due to a misconfigured setting in your `acrn.conf` file. Can you double-check the values you put, specifically for the kernel to be booted **and** the rootfs partition?

JeremyHigginsAllion, 2018-10-17:
Considering that I do not have access to the command line interface for the ACRN partition after installation, I do not know how I would be able to check the acrn.conf file.

To clarify, once I finished the Step 7 command `# systemctl start weston@cl_sos`, a Desktop appears, but it looks like it is the same Desktop that is for the `Clear Linux OS for Intel Architecture (Clear-linux-pk414-standard-4.14.52.63)`, which lives on a different partition than the ACRN Service OS.

I haven't played with that partition myself yet, but a colleague of mine said that it cannot access any of the files associated with the ACRN Service OS.

Is there a method of accessing the acrn.conf file after installation that I am unaware of?

gvancuts, 2018-10-17:
> Considering that I do not have access to the command line interface for the ACRN partition after installation, I do not know how I would be able to check the acrn.conf file.

If you press F10 when the NUC boots, do you get a menu with `UEFI: Built-in EFI Shell`?
* If so, choose that, then type the following sequence:
```
fs0:
cd EFI
cd org.clearlinux
bootloaderx64.efi
```
And that should give you the Clear Linux boot menu, and pick the `native` kernel. This will bring you to Clear Linux running bare-metal (i.e. the hypervisor is **not** running).

From there, you could be able to mount the ESP partition (e.g. `sudo mount /dev/sda1 /boot`) and inspect the `/boot/loader/entries/acrn.conf` file)

* If you don't have the option to choose `UEFI: Built-in EFI Shell`, enter the Bios (by pressing F2), go the `Advanced` menu, select `Boot`, then the `Boot Configuration` tab and select the `Internal UEFI Shell` option in the `Boot Devices` section, reboot and follow the same instructions as above.
 
> To clarify, once I finished the Step 7 command `# systemctl start weston@cl_sos`, a Desktop appears, but it looks like it is the same Desktop that is for the `Clear Linux OS for Intel Architecture [...]

Why are you saying it's living on a different partition? The `rootfs` is from the same partition and is the same, only the kernel is different.

> I haven't played with that partition myself yet, but a colleague of mine said that it cannot access any of the files associated with the ACRN Service OS.

The ACRN Service OS is your initial installation of Clear Linux that was modified, but the `rootfs` is on the same partition. The `ESP partition` may not be mounted automatically at boot time, in that case `/boot` would be empty.
 
> Is there a method of accessing the acrn.conf file after installation that I am unaware of?

What I describe above should give you access to the `acrn.conf` file. Let me know if something is not working as I described it.

mwang106, 2018-12-24:
@JeremyHigginsAllion have you resolved your issue?

Mingyuan18, 2019-03-22:
Close this issue because there is no reply for a long time

closed 2019-03-22 by Mingyuan18