projectacrn/acrn-hypervisor #1318: Kernel security check failed with variable CONFIG_VMAP_STACK

LibertadCruz, 2018-09-21:
The Clear linux in integrating the kcc tool to check for kernel security in the variables that it enables or disables. The test fails because ACRN kernel does not set  CONFIG_VMAP_STACK .

log snipet:
`sudo -E kcc /usr/lib/kernel/config-$(uname -r)' failed
# CONFIG_VMAP_STACK is not set but is required to be set to y (Guard pages for kernel stacks)

Mingyuan18, 2019-01-14:
[External_System_ID] ACRN-2469

miguelinux, 2019-03-18:
Hi,

At Clear Linux the kenrel needs to be set with security options enabled.
At ACRN project, I found that the option 

VMAP_STACK

needs to be NOT set, you can see at:

https://github.com/projectacrn/acrn-kernel/blob/master/drivers/vhm/Kconfig#L6
and
https://github.com/clearlinux-pkgs/linux-iot-lts2018/blob/master/0512-sos-add-a-config-for-VHM.patch#L87

Can you improve the code to enable VMAP_STACK?

We need to enable VMAP_STACK as it is a MANDATORY Security Settings, as it
Guard pages for kernel stacks.

miguelinux, 2019-03-18:
@shiqingg, @lyan3, @jren1, @anthonyzxu, @gvancuts, @dongyaozu 

Can you help us to solve the #1318 issue.

yakuizhao, 2019-03-18:
@miguelinux,  it is not easy to remove the !VMAP_STACK.
Currently the stack variable is used to pass the parameter to communicate with hypervisor. The hypervisor needs to read/write the corresponding variable by using the physical address. 
After the VMAP_STACK is enabled,  the stack variable can't be used. Instead the kmalloc should be used. In such case it needs a lot of code change.

yakuizhao, 2019-03-18:
@miguelinux, Do you have any estimitation date for clearlinux that the VMAP_STACK should be removed?

miguelinux, 2019-03-18:
> @miguelinux, Do you have any estimitation date for clearlinux that the VMAP_STACK should be removed?

Well, as this issue was reported a while ago, I don't have a ETA for it,
but it needs to be ASAP please.

miguelinux, 2019-03-19:
I just see the RFC, but no more discussions

1. https://lists.projectacrn.org/g/acrn-dev/message/14371?p=,,,20,0,0,0::Created,,VMAP_STACK,20,2,0,28016092
1.1 https://lists.projectacrn.org/g/acrn-dev/message/14420?p=,,,20,0,0,0::Created,,VMAP_STACK,20,2,0,28016092
2. https://lists.projectacrn.org/g/acrn-dev/message/14372?p=,,,20,0,0,0::Created,,VMAP_STACK,20,2,0,28016095
3. https://lists.projectacrn.org/g/acrn-dev/message/14373?p=,,,20,0,0,0::Created,,VMAP_STACK,20,2,0,28016097

yakuizhao, 2019-04-15:
[External_System-ID] ACRN:3391

Mingyuan18, 2019-05-10:
@LibertadCruz , This issue's patch was merged, please help to verify, if pass, please close it. Thanks!

fuzhongl, 2019-05-24:
@LibertadCruz Patch was merged. Could you please close this ticket?
Thanks!

miguelinux, 2019-05-24:
@fuzhongl  is this fix already sent to PKT team?

miguelinux, 2019-05-24:
> @fuzhongl  is this fix already sent to PKT team?
> 

Looks like not :-(

Please send the updates to the PKT team so CL can take this updates,
once the updates are in CL we can close this issue.

fuzhongl, 2019-05-24:
@miguelinux Thank you for reminding.
Will check with PKT for this.

closed 2020-04-21 by fuzhongl