projectacrn/acrn-hypervisor #1319: [APLNUC][IO][LaaG][OpenSource]SD card pass-through: UOS can’t see SD card after UOS reboot

NanlinXie, 2018-09-24:
Environment:
------------
Build link:
https://download.clearlinux.org/releases/25130/ 
Image info:
acrn-hypervisor: 2018w38.4.140000p
linux-pk414: 4.14.68 99
Hardware: APL NUC

Bug detailed description:
------------
SD card pass-through while uos reboot, UOS can not see SD card

Reproduce steps:
-------------
1:insert SD card and enable VT-d
Disk /dev/sda: 111.8 GiB, 120034123776 bytes, 234441648 sectors
/dev/sda1 2048 1048575 1046528 511M EFI System
/dev/sda2 1048576 1114111 65536 32M Linux swap
/dev/sda3 1114112 234439695 233325584 111.3G Linux root (x86-64)
Disk /dev/mmcblk0: 29.7 GiB, 31914983424 bytes, 62333952 sectors
/dev/mmcblk0p1 8192 62333951 62325760 29.7G c W95 FAT32 (LBA)

2:hide SD in sos
1): find out SD host
$lspci 
2):Find out the numeric PCI ID for SD Host
$lspci -n
3):edit parameters：
echo "10ec 5229 " > /sys/bus/pci/drivers/pci-stub/new_id
echo "0000:01:00.0" > /sys/bus/pci/devices/0000:01:00.0/driver/unbind
echo "0000:01:00.0" > /sys/bus/pci/drivers/pci-stub/bind
-s 8,passthru,0/1b/0 \
3:boot uos by pass-through SD in cmd line.
4:uos can access this SD card.
root@zcbNuc ~ # fdisk -l |grep dev
Disk /dev/vda: 8.5 GiB, 9161408512 bytes, 17893376 sectors
/dev/vda1 2048 1048575 1046528 511M EFI System
/dev/vda2 1048576 1114111 65536 32M Linux swap
/dev/vda3 1114112 17891327 16777216 8G Linux root (x86-64)
Disk /dev/mmcblk0: 29.7 GiB, 31914983424 bytes, 62333952 sectors
/dev/mmcblk0p1 8192 62333951 62325760 29.7G c W95 FAT32 (LBA)
root@zcbNuc ~ # mount /dev/mmcblk0p1 /mnt
[ 43.670565] EXT4-fs (mmcblk0p1): mounted filesystem with ordered data mode. Opts: (null)
root@zcbNuc ~ # cd /mnt
root@zcbNuc /mnt # touch ff
root@zcbNuc /mnt # ls
ff lost+found

5:reboot uos
root@zcbNuc / # reboot
[ OK ] Stopped target Host and Network Name Lookups.
Stopping Daemon for power management...
[ OK ] Stopped target Timers.

6:uos still can access the SD card after reboot
Login incorrect
zcbNuc login: root
Password:
root@zcbNuc ~ # fdisk -l |grep dev
Disk /dev/vda: 8.5 GiB, 9161408512 bytes, 17893376 sectors
/dev/vda1 2048 1048575 1046528 511M EFI System
/dev/vda2 1048576 1114111 65536 32M Linux swap
/dev/vda3 1114112 17891327 16777216 8G Linux root (x86-64)

Current result:
after reboot uos, uos can not access this SD card

Expected result:
after reboot uos, uos can see the SD card

mwang106, 2018-11-09:
[External_System_ID] ACRN-1480

Mingyuan18, 2019-03-07:
Can't reproduce this issue in below build：
ACRN TAG: 2019w09.5.140000p
SOS: 28100

closed 2019-03-07 by Mingyuan18