projectacrn/acrn-hypervisor #1363: SOS+UOS glmark2-es-wayland causes hypervisor crash

mattcurfman, 2018-09-28:
NUC7i7DNHE (Intel(R) Core(TM) i7-8650U CPU @ 1.90GHz, Mesa DRI Intel(R) UHD Graphics 620)
Clear Linux 25180
HV version 0.3-unstable-2018-09-25 10:26:50- DBG 

SOS: 1 CPU
UOS: 1 CPU

Run 'weston-es2-wayland' simultaneously in both SOS and UOS.  After 5-20 seconds, hypervisor reports 'vcpu handling pending request fail' in console and entire system hard-hangs (console frozen, SOS and UOS frozen)  100% reproducible

@hwang37

mattcurfman, 2018-09-28:
I am not able to reproduce this crash on an UP2 board

hwang37, 2018-09-29:
We'll take a look and assign to our Zhipeng.

dandanyx, 2018-10-26:
Cannot reproduce this issue on SOS #25860, please help to close it.

closed 2018-10-26 by mwang106

mattcurfman, 2018-10-26:
@dandanyx, I will re-try SOS #25860 and close if no longer reproducible. Thx!

mattcurfman, 2018-10-27:
@dandanyx, #25860 is much worse: it crashes hypervisor when booting SOS.  Since you closed this defect without waiting for my feedback, I will open a new issue.

```
ACRN Hypervisor
calibrate_tsc, tsc_khz=2700000
[7631962us][cpu=0][sev=2][seq=3]:HV version 0.3-unstable-2018-10-26 12:55:07- DBG (daily tag:) build by mockbuild, start time 7627046us
[7645087us][cpu=0][sev=2][seq=4]:API version 1.0
[7649969us][cpu=0][sev=2][seq=5]:Detect processor: Intel(R) Core(TM) i7-8650U CPU @ 1.90GHz
[7658982us][cpu=0][sev=2][seq=6]:hardware support HV
[7674659us][cpu=0][sev=3][seq=12]:System S3/S5 is NOT supported.
[7681034us][cpu=0][sev=2][seq=18]:Start VM0
[7685402us][cpu=0][sev=3][seq=19]:VMX ctrl 0x48b not fully enabled: request 0x18b but get 0x8b

ACRN:\>[17851978us][cpu=0][sev=3][seq=9600]:vlapic: Start Secondary VCPU1 for VM[0]...
[17859749us][cpu=1][sev=3][seq=9601]:VMX ctrl 0x48b not fully enabled: request 0x18b but get 0x8b

Assertion failed in file lib/memory.c,line 365 : fatal error
Host Stack: CPU_ID = 0
addr(0x153ba8)	0x0000000000160000  0x000000000007ff70  0x000000001fe00000  0x0000000000116d1c
addr(0x153bc8)	0x0000000000000000  0x0000000000153dd0  0x0000000000000000  0x0000000000160000
addr(0x153be8)	0x000000001fe00000  0x0000000000122b94  0x00000000000000ff  0x0000000000134c20
addr(0x153c08)	0x000000000129cd40  0x0000000018bedd8f  0x0010000000000028  0x0000000000153ca8
addr(0x153c28)	0x0000000000153c38  0xb3f3575f18bedd8f  0x0000000000000007  0x0000fffffffff000
addr(0x153c48)	0x0000000000134c60  0x0000000000000000  0x0000000000000001  0x0000fffffffff000
addr(0x153c68)	0x0000000264ab7df0  0x0000000001246000  0x0000000264ab7df0  0x0000000001246000
addr(0x153c88)	0x0000000264ab7df0  0x000000000000001e  0x0000000000000000  0x00000000001237a9
addr(0x153ca8)	0x0000000001246000  0x000000000011b000  0x000000000012c315  0x0000000100000000
addr(0x153cc8)	0x000000001fe00000  0xb3f3575f18bedd8f  0x0000000000000012  0x000000000124a000
addr(0x153ce8)	0x0000000080000060  0x0000000000113583  0x0000000000000096  0x0000000000000012
addr(0x153d08)	0x000000000124a000  0x0000000000000012  0x000000000000001e  0xb3f3575f18bedd8f
addr(0x153d28)	0x0000000000000000  0x000000000124a000  0x0000000c44e0b720  0x00000000c0000103
addr(0x153d48)	0x000000000000001e  0x0000000000119594  0x0000000001246000  0x000000000124e038
addr(0x153d68)	0x0000000000000001  0x0000000000000000  0x0000000000000000  0x000000000011ba5e
addr(0x153d88)	0x000000007a484730  0xb3f3575f18bedd8f  0x000000007be6d328  0x0000000000000000
```