projectacrn/acrn-hypervisor #1365: [APL][IO Mediator]can't  detect USB disk after switching the USB port of touch with build in hub and USB disk during USB mediator test

mwang106, 2018-09-29:
•Environment:
•HW: MRB D0 Premium
•Build:
SOS version:          20180925-25170
acrn-hypervisor    2018w38.4.140000p
linux-pk414             4.14.68 99
IFWI Version: ABL_1820HF1
IOC Version ：4. 0. 8
UOS kernel:     c795424a331f [upstream] drm/i915/gvt: Fix stack-out-of-bounds bug in cmd parser

•Reproduce rate:  100% 
•Reproduce steps:

1.connect touch screen with build in hub and usb2.0 disk on standard A port
2.  boot to sos
3. Launch uos by usb mediator enabled in launch script:
   -s 7,xhci,1-1:1-2:1-3:2-1:2-2:2-3:cap=apl \”
4. in aaag, check touch with build in hub and usb2.o work normally
5. switch the usb port of usb2.0 and hub
6, found only touch with build in hub can work , can't recognize usb2.0 disk.
•Expected result: touch with build in hub and usb2.0 disk work normally after switch USB port.
•Actual result: only touch with build in hub work normally . usb disk can't be recognized after step 5
•Debugging info:
console:/dev/block # xHCI: fail to find physical port 3
xHCI: do not support multiple hubs currently, reject device 1-3

mwang106, 2018-09-29:
[External_System_ID] 255282

Mingyuan18, 2018-11-20:
Not reproduce the issue on this environment:  Acrn-Hypervisor:  2018w42.5.140000p_103;  SOS: 25730 .

closed 2018-11-20 by Mingyuan18