projectacrn/acrn-hypervisor #1366: [APL][IO Mediator]can't detect all USB devices when connect usb3.0 disk on usb3.0 port during USB mediator test

mwang106, 2018-09-29:
•Environment:
•HW: MRB D0 Premium
•Build:
SOS version:          20180925-25170
acrn-hypervisor    2018w38.4.140000p
linux-pk414             4.14.68 99
IFWI Version: ABL_1820HF1
IOC Version ：4. 0. 8
UOS kernel:     c795424a331f [upstream] drm/i915/gvt: Fix stack-out-of-bounds bug in cmd parser

•Reproduce rate:  100% 
•Reproduce steps:

1.connect usb3.0 disk on usb3.0 port and usb2.0 disk on usb2.0 port
2.  boot to sos
3. Launch uos by usb mediator enabled in launch script:
   -s 7,xhci,1-1:1-2:1-3:2-1:2-2:2-3:cap=apl \”
4. in aaag, can detect both usb3.0 disk and usb2.0 disk
•Expected result: can both detect usb2.0 and usb3.0 disk in aaag with USB mediator test
•Actual result: can't detect both usb2.0 and usb3.0 disks.

mwang106, 2018-09-29:
[External_System_ID] 255285

Mingyuan18, 2018-11-20:
Not reproduce the issue on : ACRN-Hypervisor : 2018w42.5.140000p_103; SOS version: 25730

closed 2018-11-20 by Mingyuan18