projectacrn/acrn-hypervisor #1367: [APL][IO Mediator] USB Mediator DRD host mode while UOS reboot, UOS can not find the USB device on OTG port

mwang106, 2018-09-29:
Environment
HW: MRB

Image info：
SOS version： 20180914-25010
acrn-hypervisor：2018w39.2.140000p
linux-pk414：    4.14.68          98
IFWI Version：ABL_1820HF1
IOC Version ：4. 0. 8
UOS OS Type： LaaG
UOS Kernel version：c795424a331f [upstream] drm/i915/gvt: Fix stack-out-of-bounds bug in cmd parser

Reproduce rate: 100%

Reproduce steps:
1:insert USB flash disk in USB OTG port and boot sos
insert USB3.0 64GB Flash disk into Standard port;
root@sos-dom0 ~ # fdisk -l |grep dev/sd*
Disk /dev/sda: 58.1 GiB, 62411243520 bytes, 121896960 sectors
/dev/sda1        2048 121896959 121894912 58.1G 83 Linux

2:boot uos by sharing USB xHCI and DRD in cmd line
-s n,passthru,0/15/1 \
-s n,xhci,1-1:1-2:1-3:2-1:2-2:2-3:cap=apl \

3: set DRD to be host mode in uos
echo host > /sys/bus/platform/drivers/intel_xhci_usb_sw/intel_xhci_usb_sw/usb_role/intel_xhci_usb_sw-role-switch/role
cat  /sys/bus/platform/drivers/intel_xhci_usb_sw/intel_xhci_usb_sw/usb_role/intel_xhci_usb_sw-role-switch/role
role

4: uos can access USB mass storage
root@clr-a05881c790b7447cb0e8cc363174b986 ~ # fdisk -l |grep dev
GPT PMBR size mismatch (15597568 != 10239999) will be corrected by write.
Disk /dev/mmcblk0: 29.8 GiB, 32010928128 bytes, 62521344 sectors
/dev/mmcblk0p1       8192 62521343 62513152 29.8G  c W95 FAT32 (LBA)
Disk /dev/ram0: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram1: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram2: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram3: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram4: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram5: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram6: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram7: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram8: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram9: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram10: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram11: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram12: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram13: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram14: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram15: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/vda: 4.9 GiB, 5242880000 bytes, 10240000 sectors
/dev/vda1    2048   206847   204800  100M Linux filesystem
/dev/vda2  206848 10237951 10031104  4.8G Linux filesystem
Disk /dev/sda: 58.1 GiB, 62411243520 bytes, 121896960 sectors
/dev/sda1        2048 121896959 121894912 58.1G 83 Linux
Disk /dev/sdb: 14.3 GiB, 15376318464 bytes, 30031872 sectors
/dev/sdb1        2048 30031871 30029824 14.3G 83 Linux

5:uos reboot
root@clr-a05881c790b7447cb0e8cc363174b986 ~ # reboot
         Stopping Telemetrics Daemon...
         Stopping User Manager for UID 0...
         Stopping Proxy AutoConfig discovery service...

root@clr-a05881c790b7447cb0e8cc363174b986 ~ # fdisk -l |grep dev
Disk /dev/ram1: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram2: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram3: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram4: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram5: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram6: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram7: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram8: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram9: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram10: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram11: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram12: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram13: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram14: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram15: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/vda: 4.9 GiB, 5242880000 bytes, 10240000 sectors
/dev/vda1    2048   206847   204800  100M Linux filesystem
/dev/vda2  206848 10237951 10031104  4.8G Linux filesystem
Disk /dev/mmcblk0: 29.8 GiB, 32010928128 bytes, 62521344 sectors
/dev/mmcblk0p1       8192 62521343 62513152 29.8G  c W95 FAT32 (LBA)
Disk /dev/sda: 58.1 GiB, 62411243520 bytes, 121896960 sectors
/dev/sda1        2048 121896959 121894912 58.1G 83 Linux

6:echo host > /sys/bus/platform/drivers/intel_xhci_usb_sw/intel_xhci_usb_sw/usb_role/intel_xhci_usb_sw-role-switch/role
root@clr-a05881c790b7447cb0e8cc363174b986 ~ # cat /sys/bus/platform/drivers/intel_xhci_usb_sw/intel_xhci_usb_sw/usb_role/intel_xhci_usb_sw-role-switch/role
root@clr-a05881c790b7447cb0e8cc363174b986 ~ # device
root@clr-a05881c790b7447cb0e8cc363174b986 ~ # echo host > /sys/bus/platform/drivers/intel_xhci_usb_sw/intel_xhci_usb_sw/usb_role/intel_xhci_usb_sw-role-switch/role
root@clr-a05881c790b7447cb0e8cc363174b986 ~ # cat /sys/bus/platform/drivers/intel_xhci_usb_sw/intel_xhci_usb_sw/usb_role/intel_xhci_usb_sw-role-switch/role
790b7447cb0e8cc363174b986 ~ # host

7:uos still can access USB mass storage, but Actual result: UOS can not find the USB2.0 Flash disk on USB OTG port;
root@clr-a05881c790b7447cb0e8cc363174b986 ~ # fdisk -l |grep dev
GPT PMBR size mismatch (15597568 != 10239999) will be corrected by write.
Disk /dev/ram0: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram1: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram2: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram3: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram4: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram5: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram6: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram7: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram8: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram9: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram10: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram11: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram12: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram13: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram14: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/ram15: 16 MiB, 16777216 bytes, 32768 sectors
Disk /dev/vda: 4.9 GiB, 5242880000 bytes, 10240000 sectors
/dev/vda1    2048   206847   204800  100M Linux filesystem
/dev/vda2  206848 10237951 10031104  4.8G Linux filesystem
Disk /dev/mmcblk0: 29.8 GiB, 32010928128 bytes, 62521344 sectors
/dev/mmcblk0p1       8192 62521343 62513152 29.8G  c W95 FAT32 (LBA)
Disk /dev/sda: 58.1 GiB, 62411243520 bytes, 121896960 sectors
/dev/sda1        2048 121896959 121894912 58.1G 83 Linux

Expected result:
UOS can find the USB2.0 Flash disk on USB OTG port

Actual result:
UOS can not find the USB2.0 Flash disk on USB OTG port

mwang106, 2018-09-29:
[External_System_ID] 256055

Mingyuan18, 2018-11-21:
Can't reproduce this issue in this environment: ACRN-Hypervisor : 2018w42.5.140000p_103; SOS Version: 25730

closed 2018-11-21 by Mingyuan18