projectacrn/acrn-hypervisor #1371: Disable RELOC config option of hypervisor

guoch2, 2018-09-29:
Commit 60857819 Replace __emalloc() with a call to uefi allocate_page() allows UEFI FW to allocate memory for hypervisor from high to low address below 4GB.  However, this change triggers an issue (might be memory corruption), and PXE boot cannot work.  

Since root cause the issue might take some time,  the PXE boot issue blocks auto-test, we disable hypervisor relocation by default for the time being in config option, and users can enable it by themselves.  In the following weeks, if we root cause the issue, we can re-enable relocation feature.

diff --git a/hypervisor/arch/x86/Kconfig b/hypervisor/arch/x86/Kconfig
index cb0f7026..d4b41165 100644
--- a/hypervisor/arch/x86/Kconfig
+++ b/hypervisor/arch/x86/Kconfig
@@ -146,7 +146,7 @@ config MTRR_ENABLED
 
 config RELOC
        bool "Enable relocation"
-       default y
+       default n
 
 config IOMMU_INIT_BUS_LIMIT
        hex "bus limitation when iommu init"

mwang106, 2018-09-29:
[External_System_ID] ACRN-1493

Mingyuan18, 2018-11-21:
Can't reproduce this issue now.

closed 2018-11-21 by Mingyuan18