projectacrn/acrn-hypervisor #1387: Newer versions of `kconfiglib` break the documentation generation process

gvancuts, 2018-10-01:
If you install `kconfiglib` today (10/01/2018) using `pip3 install --user kconfiglib`, you will get version 10.9.1. Using that version to generate the documentation pages will lead to the following error:
```
gvancuts@gvancuts-MOBL3:~/acrn-hypervisor$ make doc
make -C /home/gvancuts/acrn-hypervisor/doc html BUILDDIR=/home/gvancuts/acrn-hypervisor/build/doc
make[1]: Entering directory '/home/gvancuts/acrn-hypervisor/doc'
Traceback (most recent call last):
  File "scripts/genrest.py", line 377, in <module>
    write_kconfig_rst()
  File "scripts/genrest.py", line 97, in write_kconfig_rst
    write_sym_rst(sym, out_dir)
  File "scripts/genrest.py", line 123, in write_sym_rst
    kconfig_definition_rst(sym))
  File "scripts/genrest.py", line 318, in kconfig_definition_rst
    for node in sc.nodes)
  File "scripts/genrest.py", line 318, in <genexpr>
    for node in sc.nodes)
  File "/home/gvancuts/.local/lib/python3.5/site-packages/kconfiglib.py", line 4989, in __str__
    return self.custom_str(standard_sc_expr_str)
  File "/home/gvancuts/.local/lib/python3.5/site-packages/kconfiglib.py", line 4998, in custom_str
    self._sym_choice_node_str(sc_expr_str_fn)
  File "/home/gvancuts/.local/lib/python3.5/site-packages/kconfiglib.py", line 5061, in _sym_choice_node_str
    indent_add_cond("default " + expr_str(default, sc_expr_str_fn),
TypeError: expr_str_rst() takes 1 positional argument but 2 were given
Makefile:48: recipe for target 'kconfig' failed
make[1]: *** [kconfig] Error 1
make[1]: Leaving directory '/home/gvancuts/acrn-hypervisor/doc'
Makefile:38: recipe for target 'doc' failed
make: *** [doc] Error 2
```

Installing an older version (such as 7.0.0: `pip3 install --user kconfiglib==7.0.0`) solves this issue.

I do not know what version broke this, I am only referencing 7.0.0 because that's on one of my other dev system where I can generate the documentation successfully.

I do not know either whether we can correctly configure the ACRN hypervisor using this latest version, this is something we should check too.

gvancuts, 2018-10-01:
> I do not know either whether we can correctly configure the ACRN hypervisor using this latest version, this is something we should check too.

Just ran a quick test and the Kconfig stuff in the hypervisor code seems to work just fine with the latest `kconfiglib` version.

So either we (I guess I'm looking at @dbkinder here :-)) fix our doc generation tools so it works with the latest version, or we specify a version that's known to work in our Getting Started Guide.

dbkinder, 2018-10-01:
so it works fine for you with the latest kconfiglib version (10.9.1)?

dbkinder, 2018-10-01:
Interesting.  I have a really old kconfiglib 6.1.1 installed on my ubuntu box.  I updated it to 10.9.1 and I see the failure now too.  The latest genrest.py from the Zephyr project works with the latest kconfiglib so I'll submit a PR to use this genrest.py for ACRN too.

dbkinder, 2018-10-01:
... but now I see the CI `make docs` is failing because it's not responding to changes in the requirements.txt file.  Sigh.

ulfalizer, 2018-10-02:
@dbkinder @gvancuts 

Sorry about that!

This is fallout from adding a less hacky way to customize expression printing (to turn symbols into links, in this case), which accidentally broke the old hacky way by adding a parameter to `Kconfig.expr_str()` (for passing a custom symbol printing function).

Another change might be needed too. In `acrn-hypervisor/doc/Makefile`, Instead of

```make
kconfig:
	$(Q)srctree=../hypervisor \
	    python3 scripts/genrest.py ../hypervisor/Kconfig reference/kconfig/
```

make it

```make
kconfig:
	$(Q)srctree=../hypervisor \
	    python3 scripts/genrest.py Kconfig reference/kconfig/
```

Previously, the top-level Kconfig passed to `Kconfig.__init__()` wasn't looked up relative to `srctree` (but all other Kconfig files were). That was a pretty major design oversight, because it meant that scripts needed to take `srctree` into account to work correctly out-of-tree.

It seemed bad enough to warrant breaking backwards compatibility to fix it. With all Kconfig files consistently looked up relative to `srctree`, everything just works out-of-tree. You get a hint in the error message now if the top-level Kconfig file can't be found and `srctree` is set.

Kconfiglib uses [semantic versioning](https://semver.org/) by the way, so if you want to be absolutely sure that nothing ever breaks, you can lock to the major version (10 at the moment). If you search for "Semantic versioning" in the [Kconfiglib README](https://github.com/ulfalizer/kconfiglib), you'll find a list of all API changes between major versions (most of it is minor stuff).

I suspect I've vetted out the really bad stuff by now though, and I don't break backwards compatibility otherwise. It's mostly been some baby issues with the new Kconfiglib 2 API. Sorry about that!

gvancuts, 2018-10-02:
> Sorry about that!

No problem

> 
> This is fallout from adding a less hacky way to customize expression printing (to turn symbols into links, in this case), which accidentally broke the old hacky way by adding a parameter to `Kconfig.expr_str()` (for passing a custom symbol printing function). Didn't help that the parameter has a default value for that particular hack.
> 
> Another change might be needed too. In `acrn-hypervisor/doc/Makefile`, Instead of
> 
> ```makefile
> kconfig:
> 	$(Q)srctree=../hypervisor \
> 	    python3 scripts/genrest.py ../hypervisor/Kconfig reference/kconfig/
> ```
> 
> make it
> 
> ```makefile
> kconfig:
> 	$(Q)srctree=../hypervisor \
> 	    python3 scripts/genrest.py Kconfig reference/kconfig/
> ```

@dbkinder , I've tested this change in my environment and it works. It sounds like a good idea to adopt it in your patch but I'll let you be the judge on this.

> 
> Previously, the top-level Kconfig passed to `Kconfig.__init__()` wasn't looked up relative to `srctree` (but all other Kconfig files were). That was a pretty major design oversight, because it meant that all scripts needed to take `srctree` into account to work correctly out-of-tree.
> 
> It seemed bad enough to warrant breaking backwards compatibility to fix it. With all Kconfig files consistently looked up relative to `srctree`, everything just works out-of-tree. You get a hint in the error message now if the top-level Kconfig file can't be found and `srctree` is set.
> 
> Kconfiglib uses [semantic versioning](https://semver.org/) by the way, so if you want to be absolutely sure that nothing ever breaks, you can lock to the major version (10 at the moment). If you search for "Semantic versioning" in the [Kconfiglib README](https://github.com/ulfalizer/kconfiglib), you'll find a list of all API changes between major versions (most of it is minor stuff).
> 

So perhaps we can relax a bit the version specified in `doc/scripts/requirements.txt` and use `kconfiglib==10.*` instead? (I just tried that, it installed 10.9.2 on my system this morning (instead of 10.9.1 yesterday) and the documentation generation worked like a charm with that version too.

> I suspect I've vetted out the really bad stuff by now though, and I don't break backwards compatibility otherwise. It's mostly been some baby issues with the new Kconfiglib 2 API. Sorry about that!

No worries, and thanks for all your input and reactivity, it's much appreciated!