projectacrn/acrn-hypervisor #1389: CI 'make docs' needs to honor requirements.txt changes

dbkinder, 2018-10-01:
The document build requires specific versions of python-based doc generation tools specified in the doc/scripts/requirements.txt file.  The CI environment needs to respond to changes to this file, for example, by running:

```
   pip3 install --user -r scripts/requirements.txt
```
as part of the workspace environment setup for running the doc build.  We really can't add this to the doc build process itself (`make html`) since that would unexpectedly change the user's execution environment. 

If the `genrest.py` script (part of the doc build) is run with an out-of-sync version of kconfiglib, errors such as this can show up:

```
run make doc in acrn-hypervisor
make -C /workspace/acrn-hypervisor/doc html BUILDDIR=/workspace/acrn-hypervisor/build/doc
make[1]: Entering directory '/workspace/acrn-hypervisor/doc'
Traceback (most recent call last):
  File "scripts/genrest.py", line 441, in <module>
    write_kconfig_rst()
  File "scripts/genrest.py", line 94, in write_kconfig_rst
    for sym in sorted(kconf.unique_defined_syms, key=lambda sym: sym.name):
AttributeError: 'Kconfig' object has no attribute 'unique_defined_syms'
make[1]: *** [Makefile:48: kconfig] Error 1
make[1]: Leaving directory '/workspace/acrn-hypervisor/doc'
make: *** [Makefile:38: doc] Error 2
Build step 'Execute shell' marked build as failure
```

dbkinder, 2018-10-10:
@NanlinXie Basically, we need the CI script that runs a ``make doc`` to first do a ``pip3 install --user -r scripts/requirements.txt`` in case this file was changed (which it has).  The CI ``make doc`` test is currently failing because of this.

lijinxia, 2018-10-10:
I met a similar problem with my local docker image, clearlinux:25020, after installing the packages with your requirements.txt

Traceback (most recent call last):
  File "/usr/sbin/sphinx-build", line 11, in <module>
    load_entry_point('Sphinx==1.8.0', 'console_scripts', 'sphinx-build')()
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 484, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2707, in load_entry_point
    return ep.load()
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2325, in load
    return self.resolve()
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2331, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/lib/python3.7/site-packages/sphinx/cmd/build.py", line 25, in <module>
    from sphinx.application import Sphinx
  File "/usr/lib/python3.7/site-packages/sphinx/application.py", line 29, in <module>
    from sphinx.config import Config, check_unicode
  File "/usr/lib/python3.7/site-packages/sphinx/config.py", line 18, in <module>
    from typing import Any, NamedTuple, Union
  File "/usr/lib/python3.7/site-packages/typing.py", line 1356, in <module>
    class Callable(extra=collections_abc.Callable, metaclass=CallableMeta):
  File "/usr/lib/python3.7/site-packages/typing.py", line 1004, in __new__
    self._abc_registry = extra._abc_registry
AttributeError: type object 'Callable' has no attribute '_abc_registry'

After upgrade sphinx from 1.7.7 to 1.8.0, the issue can be fixed. I wonder if the package must be the specific version or minimum... @dbkinder

lijinxia, 2018-10-10:
yes, I tried with my older docker image(22530), it did work, but failed with my latest image(25020).
I mean my other python packages may have dependency with sphinx up to 1.8.0 as shown in the Traceback log :(

gvancuts, 2018-10-10:
> yes, I tried with my older docker image(22530), it did work, but failed with my latest image(25020).
> I mean my other python packages may have dependency with sphinx up to 1.8.0 as shown in the Traceback log :(

@NanlinXie , is the traceback from running `make doc` or from another script that you run as part of CI

lijinxia, 2018-10-10:
@gvancuts when running` make html`. It was tested with my local images, not deployed on CI system yet!
I found my latest python installed was python3.7, while in my older image, it was python3.6

> @NanlinXie , is the traceback from running `make doc` or from another script that you run as part of CI

@gvancuts when running` make html`. It was tested with my local images, not deployed on CI system yet!
I found my latest python installed was python3.7, while in my older image, it was python3.6

gvancuts, 2018-10-10:
I can reproduce this. I was thinking that upgrading all Python deps [1] may just be a good option but not so much... I'm getting new/different errors both on Fedora 27 [2] and Clear Linux 25400 [3] :(

[1] `pip3 install --user -U breathe sphinx docutils sphinx-rtd-theme kconfiglib`
```
Listing versions of doc build dependencies found on your system...

  breathe                   version: 4.10.0
  sphinx                    version: 1.8.1
  docutils                  version: 0.14
  sphinx-rtd-theme          version: 0.4.2
  kconfiglib                version: 10.12.0
  doxygen                   version: 1.8.13

```

[2] Errors with latest packages on Fedora 27
```
make doc
make -C /home/gvancuts/work/acrn/acrn-hypervisor/doc html BUILDDIR=/home/gvancuts/work/acrn/acrn-hypervisor/build/doc
make[1]: Entering directory '/home/gvancuts/work/acrn/acrn-hypervisor/doc'

New errors/warnings found, please fix them:
==============================================

/home/gvancuts/.local/lib/python3.6/site-packages/sphinx/application.py:413: RemovedInSphinx20Warning: app.verbose() is now deprecated. Use sphinx.util.logging instead.
  RemovedInSphinx20Warning)
/home/gvancuts/.local/lib/python3.6/site-packages/sphinx/ext/autodoc/__init__.py:136: RemovedInSphinx20Warning: AutodocReporter is now deprecated. Use sphinx.util.docutils.switch_source_input() instead.
  RemovedInSphinx20Warning)
/home/gvancuts/work/acrn/acrn-hypervisor/doc/api/hypercall_api.rst:12: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]
  
  ^
/home/gvancuts/work/acrn/acrn-hypervisor/doc/api/hypercall_api.rst:12: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]
  
  ^

```

[3] Errors with latest Python deps on Clear Linux 25400
```
make doc
make -C /home/cl_sos/acrn-hypervisor/doc html BUILDDIR=/home/cl_sos/acrn-hypervisor/build/doc
make[1]: Entering directory '/home/cl_sos/acrn-hypervisor/doc'

New errors/warnings found, please fix them:
==============================================

/home/cl_sos/.local/lib/python3.7/site-packages/sphinx/application.py:413: RemovedInSphinx20Warning: app.verbose() is now deprecated. Use sphinx.util.logging instead.
  RemovedInSphinx20Warning)
/home/cl_sos/.local/lib/python3.7/site-packages/sphinx/ext/autodoc/__init__.py:136: RemovedInSphinx20Warning: AutodocReporter is now deprecated. Use sphinx.util.docutils.switch_source_input() instead.
  RemovedInSphinx20Warning)
/home/cl_sos/acrn-hypervisor/doc/api/hypercall_api.rst:12: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]
  
  ^
/home/cl_sos/acrn-hypervisor/doc/api/hypercall_api.rst:12: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]
  
  ^
WARNING: dot code 'digraph G {\n   rankdir=LR;\n   bgcolor="transparent";\n   UEFI -> "acrn.efi" -> "OS\\nBootloader" ->\n     "SOS\\nKernel" -> "ACRN\\nDevice Model" -> "Virtual\\nBootloader";\n}\n': dot exited with error:
[stderr]
b'Format: "png" not recognized. Use one of: canon cmap cmapx cmapx_np dot dot_json eps fig gv imap imap_np ismap json json0 mp pic plain plain-ext pov ps ps2 svg svgz tk vml vmlz xdot xdot1.2 xdot1.4 xdot_json\n'
[stdout]
b''
WARNING: dot code 'digraph {\n   "a" -> "b"\n}': dot exited with error:
[stderr]
b'Format: "png" not recognized. Use one of: canon cmap cmapx cmapx_np dot dot_json eps fig gv imap imap_np ismap json json0 mp pic plain plain-ext pov ps ps2 svg svgz tk vml vmlz xdot xdot1.2 xdot1.4 xdot_json\n'
[stdout]
b''
 <snip>
```

NanlinXie, 2018-10-19:
@gvancuts Any update? When we used the latest clearlinux as docker image, it failed to run "make doc". Could this issue be solved?

dbkinder, 2018-10-22:
@gvancuts @NanlinXie @lijinxia 

For now, we should not be using Clear Linux for the doc build environment.  I've only tested doc building and tool dependencies on Ubuntu (but I suspect Fedora is OK too). 

Please use the versions of pip3 (python) components listed in requirements.txt -- those have been tested and known to work together.  In particular, Sphinx 1.7.7/Breathe 4.9.1/docutils/0.14 work together. 

There are issues with Sphinx 1.8.0 and later that introduced changes that break the doc build or produce unexpected errors or warnings (as you've seen).

gvancuts, 2018-10-23:
This is a problem we need to fix @dbkinder . The official build environment for ACRN is Clear Linux, it's not a good approach to recommend to people that they use 2 different systems, one to build the binaries and the other to build the documentation.

hwang37, 2018-10-23:
Agree with Geoffroy. Let's try Clear Linux as doc build environment. 
@xinyunliu our developer to help the env set up if need any debug work.

dbkinder, 2018-10-23:
I set up a CL system and added the bundles mention in the ACRN getting started guide.  Those instructions didn't add ``make``, so I also added the ``os-core-dev`` bundle.  I cloned the acrn-hypervisor and acrn-kernel repos (as documented in the doc buiding tutorial on our website), and did a pip3 install --user (also as documented).  I added ~/.local/bin to the FRONT of the $PATH environment variable, and successfully built the documentation with a ``make html``. 

So, if you use the python package versions listed in the doc/scripts/requirements.txt on a CL system, you can successfully build the documentation.  You'll get errors using Sphinx 1.8.1, so don't do that.

xinyunliu, 2018-10-26:
in CL 22530 environment, I have similar issue.
```
Package                      Version
---------------------------- ---------
docutils                     0.14
kconfiglib                   9.0.0
Sphinx                       1.7.4
```
log is similar to what @dbkinder got:
``` syslog
$ make html
Traceback (most recent call last):
  File "scripts/genrest.py", line 441, in <module>
    write_kconfig_rst()
  File "scripts/genrest.py", line 87, in write_kconfig_rst
    kconf = kconfiglib.Kconfig(sys.argv[1])
  File "/usr/lib/python3.6/site-packages/kconfiglib.py", line 742, in __init__
    self._file = self._open_enc(filename, _UNIVERSAL_NEWLINES_MODE)
  File "/usr/lib/python3.6/site-packages/kconfiglib.py", line 2947, in _open_enc
    open(filename, mode, encoding=self._encoding)
FileNotFoundError: [Errno 2] No such file or directory: 'Kconfig'
```
After upgrade the related package via `pip3 install --user -U breathe sphinx docutils sphinx-rtd-theme kconfiglib`, Can generate documents via `make html`  without error.

```
breathe                      4.10.0
docutils                     0.14
Sphinx                       1.8.1
sphinx-rtd-theme             0.4.2
kconfiglib                   10.12.0
```

gvancuts, 2018-10-30:
I'm not seeing any error anymore in the following environment:
```
gvancuts@clr-geoffroy ~/work/acrn/acrn-hypervisor $ swupd info
Installed version: 25620
Version URL:       https://download.clearlinux.org/update/
Content URL:       https://cdn.download.clearlinux.org/update/

Listing versions of doc build dependencies found on your system...

  breathe                   version: 4.9.1
  sphinx                    version: 1.7.7
  docutils                  version: 0.14
  sphinx-rtd-theme          version: 0.4.0
  kconfiglib                version: 10.17.0
  doxygen                   version: 1.8.14

gvancuts@clr-geoffroy ~/work/acrn/acrn-hypervisor $ python3 --version
Python 3.7.0
```
@NanlinXie , are you still seeing this?

lijinxia, 2018-11-05:
@gvancuts I setup CL docker, the same as your version, but get the same error as before.
`[lijinxia@clearlinux_acrn:25620 acrn-hypervisor]$swupd info
Installed version: 25620
Version URL:       https://download.clearlinux.org/update/
Content URL:       https://cdn.download.clearlinux.org/update/
[lijinxia@clearlinux_acrn:25620 acrn-hypervisor]$doc/scripts/show-versions.py
Listing versions of doc build dependencies found on your system...

  breathe                   version: 4.9.1
  sphinx                    version: 1.7.7
  docutils                  version: 0.14
  sphinx-rtd-theme          version: 0.4.0
  kconfiglib                version: 10.17.0
  doxygen                   version: 1.8.14
`
The Traceback
`make[1]: [Makefile:56: html] Error 1 (ignored)

New errors/warnings found, please fix them:
==============================================

Traceback (most recent call last):
  File "/usr/sbin/sphinx-build", line 11, in <module>
    load_entry_point('Sphinx==1.8.1', 'console_scripts', 'sphinx-build')()
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 484, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2714, in load_entry_point
    return ep.load()
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2332, in load
    return self.resolve()
  File "/usr/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2338, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/lib/python3.7/site-packages/sphinx/cmd/build.py", line 25, in <module>
    from sphinx.application import Sphinx
  File "/usr/lib/python3.7/site-packages/sphinx/application.py", line 29, in <module>
    from sphinx.config import Config, check_unicode
  File "/usr/lib/python3.7/site-packages/sphinx/config.py", line 18, in <module>
    from typing import Any, NamedTuple, Union
  File "/usr/lib/python3.7/site-packages/typing.py", line 1356, in <module>
    class Callable(extra=collections_abc.Callable, metaclass=CallableMeta):
  File "/usr/lib/python3.7/site-packages/typing.py", line 1004, in __new__
    self._abc_registry = extra._abc_registry
AttributeError: type object 'Callable' has no attribute '_abc_registry'
make[1]: *** [Makefile:57: html] Error 1
make[1]: Leaving directory '/home/lijinxia/docker_22530/EB_0813/acrn-hypervisor/doc'
make: *** [Makefile:38: doc] Error 2
`
and my typing package info
`Name: typing
Version: 3.6.6
Summary: Type Hints for Python
Home-page: https://docs.python.org/3/library/typing.html
Author: Guido van Rossum, Jukka Lehtosalo, Łukasz Langa, Ivan Levkivskyi
Author-email: jukka.lehtosalo@iki.fi
License: PSF
Location: /usr/lib/python3.7/site-packages
Requires:
Required-by:
`

lijinxia, 2018-11-13:
> It looks like you also have a sphinx 1.8.1 globally installed, and it’s trying to use it (and not the 1.7.7 installed for the user).

Exactly! Uninstall the sphinx 1.8.1 globally and re-install 1.7.7, it works now! 👍

dbkinder, 2019-02-28:
This appears to be resolved now, yea.

closed 2019-02-28 by dbkinder