projectacrn/acrn-hypervisor #1402: [APL][Hypervisor] guest can not recovery successfully after kernel panic. [Assertion `irq_counts[irq] == 0 || irq_counts[irq] == IRQ_DISABLED' failed.]

mwang106, 2018-10-08:
1. Environment
--------------------------------------------------------------
[Board]: D0 MRB
HW: MRB D0 Premium
Kernel Version/Android Version4.14.69-quilt-2e5dc0ac-g9445f3daf16e/9
AOSP VersionPPR1.180610.011
ABL IFWI Versionrel.1820-Android-HF1
IOC Version4.0.8
ACRN-Hypervisor 2018w39.2.140000p_92
SOS Build Link20180926-25180
SOS VersionInstalled version: 25180
SOS Kernel Version4.14.69-102.pk414-sos #1 SMP PREEMPT Tue Sep 25
cAVS FW Version9.22.1.3472
UOS OS Type: AaaG
UOS Kernel version: 4.14.69-quilt-2e5dc0ac-g9445f3daf16e

2.  Steps
--------------------------------------------------------------
Reproduce step:
1. Launch SOS
2. Use acrn-dm to launch android with -vsbl <valiad vsbl image path >
(./launch_UOS.sh -V 2)
3. Trigger kernel panic
      echo 1 > /proc/sys/kernel/sysrq
      echo c > /proc/sysrq-trigger
4. repeat step3 5 times
5. Observe AaaG guest can recovery successfully every time after kernel panic. 

3. Current result:
--------------------------------------------------------------
step5: AaaG can not recovery successfully at the second times, and itshows “acrn-dm: hw/pci/irq.c:110: pci_irq_reserve: Assertion `irq_counts[irq] == 0 || irq_counts[irq] == IRQ_DISABLED' failed.”

mwang106, 2018-10-08:
[External_System_ID] 256887

Mingyuan18, 2018-11-21:
Can't reproduce this issue in below environment: ACRN-Hypervisor 2018w43.5.140000p_106; SOS Version:25920

closed 2018-11-21 by Mingyuan18