projectacrn/acrn-hypervisor #1410: [APL][IO Mediator][LaaG]Voice Assistant is not working

lishuangx, 2018-10-09:
Environment:
AaaG weekly build:
BW39C Internalhttps://mcg-depot.intel.com/artifactory/cactus-absp-jf/build/eng-builds/master/PSI/weekly/2018_WW39_C/ 
Release Note:Kernel Version/Android Version4.14.69-quilt-2e5dc0ac-g9445f3daf16e/9
AOSP VersionPPR1.180610.011
ABL IFWI Versionrel.1820-Android-HF1
IOC Version4.0.8
ACRN-Hypervisor 2018w39.2.140000p_92
SOS Build Link20180926-25180
SOS VersionInstalled version: 25180
SOS Kernel Version4.14.69-102.pk414-sos #1 SMP PREEMPT Tue Sep 25
cAVS FW Version9.22.1.3472
Reproduce rate: always
step:
1 . Boot acrn image
2 . flash steeringWheel.rcb in to CAN box
3 . Press 'A' on CAN box
Expected test result:
--------------------------------
Voice assistant UI should pop up
Actual test result:
---------------------------
It doesn't show any voice assistant UI

Additional Information:
-------------------------------------
while grep signal id "logcat | grep -i signalid" -> not able to see 722

lishuangx, 2018-10-09:
[External_System_ID] 258632

Mingyuan18, 2018-12-21:
Can't reproduce this issue in below build:
ACRN-Hypervisor : 2018w42.5.140000p_103
SOS : 25730

closed 2018-12-21 by Mingyuan18