projectacrn/acrn-hypervisor #1422: dm: clean up virtio-block and block_if

peterfang, 2018-10-10:
1. Remove some dead code from Bhyve
2. Add debugging info
3. Add I/O range checks

hwang37, 2018-10-10:
@conghuic23  to co-review these code changes.
@peterfang  Is this a new feature? I change the type to feature if yes and you can use it for "Tracked-On". @junjiemao1 to add bi-direction link for new feature

peterfang, 2018-10-10:
> @conghuic23 to co-review these code changes.
> @peterfang Is this a new feature? I change the type to feature if yes and you can use it for "Tracked-On". @junjiemao1 to add bi-direction link for new feature

The changes are cleanups and minor improvements. I think you can put this down as a bug.

junjunshan1, 2018-10-10:
[External_System_ID] 258897

Mingyuan18, 2018-12-19:
No regression issue found.

closed 2018-12-19 by Mingyuan18