projectacrn/acrn-hypervisor #1434: [APL][IO Mediator][AaaG]USB mediator mode does not support multi-layer hub

lidongwangx, 2018-10-11:
Environment: version release:
SOS Kernel version:4.14
UOS OS Type: AaaG
UOS Kernel version: 4.14
Reproduce rate: always
Image info:
SOS version:           20180926-25180
acrn-hypervisor:   2018w39.2.140000p
linux-pk414:             4.14.69        102
IFWI Version: ABL_1820HF1
IOC Version ：4. 0. 8

step:
1.insert a hub to MRB USB port
2.insert another hub to first hub's USB port
3.insert USB 3.0 disk to hub
4.add "-s n ,xhci,1-1:1-2:1-3:2-1:2-2:2-3 \" to scripts
5.boot uos

Expected result:
uos can get USB disk
Actual result:
uos can not get USB disk

lidongwangx, 2018-10-11:
[External_System_ID] 258441

Mingyuan18, 2018-12-21:
Can't reproduce this issue in below build:
ACRN-Hypervisor : 2018w43.5.140000p_106
SOS : 25920

closed 2018-12-21 by Mingyuan18