projectacrn/acrn-hypervisor #1449: [APL][IO Mediator][AaaG] boot uos when test virtio_input  then return error: read from STATUS: bad size 1

lidongwangx, 2018-10-12:
Environment:
HW: MRB D0 Premium
MRB Daily test(E2E-BKC Build):
SOS Kernel version:4.19
UOS OS Type: Android
UOS Kernel version: 4.19
Reproduce rate: always
Reproduce steps:
1.  plug in USB touch with USB3.0 port and make sure virtio-input enabled in kernel config
CONFIG_VIRTIO_INPUT=y
2.  Remove USB related pass thru/mediator parameters in launch_uos.sh
3.  check input device info. And find the input event in SOS:
           $ cat /proc/bus/input/devices
           $ ls -l /dev/input/by-id
4.  add virtio-input parameter in launch_uos.sh:
           -s 7,virtio-input,/dev/input/event1
5.  boot up uos and check touch take effect on uos

Expected result:
  touch works normally in uos(AaaG)
Actual result:
boot uos then return error: read from STATUS: bad size 1

lidongwangx, 2018-10-12:
[External_System_ID] 259262

jingrongx, 2018-10-18:
The Issue has reproduced.

Build:
MRB Daily test(E2E-BKC #2 Build): 
Image info:
Hypervisor: acrn-2018w41.4-140000p
Hypervisor patch: https://github.com/minhe1/acrn-hypervisor/commit/c4090f4c7d2046c9a990a1ca321a36b7dcd47e44.patch (pvmmio)
https://github.com/yakuizhao/acrn-hypervisor/commit/5c960a86cf6fc9c8f2eb33dfa0f5c86976efdb90.patch (Revert "DM: Disable plane_restriction on 4.19 kernel)
Kernel branch: acrn_mainline_v419-rc6_ww413
Clearlinux: 20180926-25180

Print much message like this:
virtio_input: read from STATUS: bad size 1

jingrongx, 2018-10-23:
The issue has been fixed in AaaG's ww43.2  4.19 E2Eversions.
Build:
Use 4.19 E2E Build

Mingyuan18, 2018-11-21:
Can't reproduce this issue in this environment: ACRN-Hypervisor : 2018w42.5.140000p_103; SOS: 25730

closed 2018-11-21 by Mingyuan18