projectacrn/acrn-hypervisor #1466: [MRB][HV][SIT] S3 doesn’t take effect and no display when test S3

lishuangx, 2018-10-15:
Follow s3 operation steps to “acrnd &”, Andriod did not start ,screen no display and S3 doesn’t take effect .
1. environment and steps
hardware: MRB 8G
Build: ww41.5_B2
Build Link: 
>Link option#1: http://10.239.147.147/acrn_daily_build/ww41.5/MRB/B2
IOC: 4.0.8
ABL:1820HF1

2. steps:
1) BOOT UP BOARD
2) Launch sos
3). In SOS, copy launch_uos.sh from /usr/share/acrn/samples/apl-mrb/ to ~/;
4). Edit launch_uos.sh, remove the line include “virtio-hyper_dmabuf” to disable hyperdma; (2 lines total, one for AaaG and one for LaaG)
5). Open another ssh section to SOS, input: “acrnctl add ./launch_uos.sh -V 2” (note: -V 1 for LaaG, 2 for AaaG),
you will get output as below:
“vm1-xx:yy:zz added”
Note: this only need run once, next time you might get “vm1 … already exist …”,
6). Input “acrnctl list” to confirm VM name(i.e. vm1-xx:yy:zz);
7). Run “acrnd &” command in the SOS
8). Press ignition key to trigger/resume system S3;

3. Observations
Android did not starxt ,screen no display.

4. logs

lishuangx, 2018-10-15:
[External_System_ID] ACRN-1558

Mingyuan18, 2018-12-24:
No regression issue in test.

closed 2018-12-24 by Mingyuan18