projectacrn/acrn-hypervisor #1476: [APL][HV][1/10]SOS will crash after manual triggle kernel panic

mwang106, 2018-10-16:
1.Environment
MRB image:
MRB Daily test(E2E-BKC Build):

Hypervisor: acrn-2018w41.4-140000p
sos kernel: Linux sos-dom0 4.19.0-rc6-quilt-2e5dc0ac-gc0a146577396 
SOS Version: 25180
IFWI Version: ABL_1820HF1
IOC Version ：4. 0. 8

2. Steps
1. Launch SOS
2. Use acrn-dm to launch android with -vsbl <valiad vsbl image path >
(./launch_UOS.sh -V 2)
3. Trigger kernel panic
echo 1 > /proc/sys/kernel/sysrq
echo c > /proc/sysrq-trigger

3. Current result:
SOS will crash after manual triggle kernel panic.

4. Expected result:
SOS and UOS recovery successfully after triggle kernel panic.
5. repetition rate
10%

Notes: Ping SOS failed.

mwang106, 2018-10-16:
[External_System_ID] ACRN-1551

Mingyuan18, 2018-11-21:
Can't reproduce this issue in this environment: ACRN-Hypervisor : 2018w43.5.140000p_106; SOS Version:25920

closed 2018-11-21 by Mingyuan18