projectacrn/acrn-hypervisor #1484: Creating a new default configuration does not work

gvancuts, 2018-10-16:
We have instructions on how to create a new default configuration [here](https://projectacrn.github.io/latest/getting-started/apl-nuc.html#create-a-new-default-configuration). However, these instructions do not work. Here is how to quickly reproduce the error:
```
$ cd acrn-hypervisor/hypervisor/arch/x86/configs
$ cp uefi.config test.config
$ cd acrn-hypervisor/hypervisor
$ make PLATFORM=test
```
This will lead to the following error:
```
make: *** No rule to make target '/home/gvancuts/work/acrn/acrn-hypervisor/hypervisor/build/bsp/test/test.o', needed by '/home/gvancuts/work/acrn/acrn-hypervisor/hypervisor/build/acrn.out'.  Stop.
```

gvancuts, 2018-10-16:
I'm not sure I'm understanding how this feature is supposed to work. I was expecting that one could do a `make PLATFORM=new-platform` from either the top-level directory or within the `hypervisor/` folder and it would generate the hypervisor for that configuration. **But**, that is not the case (and I find this confusing).

What I was expecting to be able to do:
```
$ cd hypervisor
$ make PLATFORM=new-platform
```
That fails because of some path that are wrong. The patch I proposed in #1485 fixes the paths but what it does not **realy** compile ACRN for the `PLATFORM` specified.

What works:
```
$ cd hypervisor
$ make defconfig PLATFORM=new-platform
$ make
```
I find this rather confusing.

mwang106, 2018-12-24:
[https://github.com/projectacrn/acrn-hypervisor/pull/1594](url) this issue.

closed 2018-12-24 by mwang106