projectacrn/acrn-hypervisor #1557: doc: Update ACRN-specific coding style exceptions with regard to Linux style

junjiemao1, 2018-10-23:
ACRN mostly follows Linux coding style, but with a few exceptions for, e.g. MISRA-C compliance. The checkpatch.pl script shall be customized before applying to the ACRN code so that warnings on such exceptions are suppressed.

dbkinder, 2019-02-28:
@junjiemao1 @NanlinXie @lijinxia @hwang37 Can we close this issue as fixed now?

eternalNight, 2019-03-01:
@dbkinder I think so. The customized configuration to checkpatch has been merged months ago.

@NanlinXie Would you please verify and close this issue? Thanks.

junjiemao1, 2019-03-01:
@dbkinder I think so. The customized configuration to checkpatch has been merged months ago.

@NanlinXie Would you please verify and close this issue? Thanks.

dbkinder, 2019-03-22:
We're also maintaining a [coding style doc](https://projectacrn.github.io/latest/developer-guides/coding_guidelines.html) now, so the issue is resolved.

closed 2019-03-22 by dbkinder