projectacrn/acrn-hypervisor #1566: [APL][IO Mediator][AaaG] USB Device shared by XHCI, When boot up UOS, plugged in the USB Device will appear Crash

lishuangx, 2018-10-23:
Environment
HW: MRB

build:
MRB Daily test(E2E-BKC#4  Build):  Please verify know issue first, then execute the daily test. Thanks!
AaaG:  https://buildbot.sh.intel.com/absp/builders/master-engineering/builds/79068
SOS Image: http://10.239.147.147/temp/E2E-BKC/20181016/
Build Info:
Hypervisor: https://github.com/projectacrn/acrn-hypervisor.git  (including 259736 fix + reenable audio in laucnh_uos )
CommitID: c6c1e42 HV: fix 'missing for discarded return value' violations
Hypervisor patch: https://github.com/projectacrn/acrn-hypervisor/pull/1474.patch (Jason’s patch for vuart)
Kernel: https://github.com/projectacrn/acrn-kernel.git
Commit ID: c02043a  Enable CONFIG_USB_ROLES_INTEL_XHCI for USB
Kernel Branch: acrn_mainline_v419-rc7_ww421
Clearlinux:25180
 
Image info：
SOS version： 20181016-25180
acrn-hypervisor：acrn-2018w42.1-140000f
linux-pk414：Linux sos-dom0 4.19.0-rc7-quilt-2e5dc0ac-gc02043aef573 #1 SMP PREEMPT Tue Oct 16 05:46:04 UTC 2018 x86_64 GNU/Linux
IOC Version ：4.19.0-rc7-quilt-2e5dc0ac-g3d0e8f0e5716
UOS OS Type： AaaG
UOS Kernel version：Linux localhost 4.19.0-rc7-quilt-2e5dc0ac-g3d0e8f0e5716 #1 SMP PREEMPT Mon Oct 15 12:04:51 UTC 2018 x86_64

Reproduce rate: 100%

Reproduce steps:

1:insert a Hub into Standard 3.0 port

2:boot uos by sharing USB xHCI and DRD in cmd line
-s n,passthru,0/15/1 \
-s n,xhci,1-1:1-2:1-3:2-1:2-2:2-3 \

3: in the process of booting uos, plugged in the USB flash disk on HUB;

Expected result:
can  boot up UOS successfully

Actual result:
UOS boot up fail,appear crash

Debugging info:
please see the attachment.

Attachement:
please see the attachment.

lishuangx, 2018-10-23:
[External_System_ID] 260005

Mingyuan18, 2018-11-21:
Can't reproduce this issue in this environment: ACRN-Hypervisor : 2018w45.5.140000p_116; SOS version: 25660

closed 2018-11-21 by Mingyuan18