projectacrn/acrn-hypervisor #1567: [APL][IO Mediator][AaaG]usb_dev_comp_req: Assertion after testing USB camera

lishuangx, 2018-10-23:
Environment: version release:
SOS Kernel version:4.14
UOS OS Type: AaaG
UOS Kernel version: 4.14
Reproduce rate: always

BW39C Internalhttps://mcg-depot.intel.com/artifactory/cactus-absp-jf/build/eng-builds/master/PSI/weekly/2018_WW39_C/ 
Release Note:Kernel Version/Android Version4.14.69-quilt-2e5dc0ac-g9445f3daf16e/9
AOSP VersionPPR1.180610.011
ABL IFWI Versionrel.1820-Android-HF1
IOC Version4.0.8
ACRN-Hypervisor 2018w39.2.140000p_92
SOS Build Link20180926-25180
SOS VersionInstalled version: 25180
SOS Kernel Version4.14.69-102.pk414-sos #1 SMP PREEMPT Tue Sep 25
cAVS FW Version9.22.1.3472

step:
1.insert usb camera in USB port
2.start UOS by lanuch_uos.sh -V 2
3.executed "  ./yavta -u -c1 -n1 -F -s 1920x1080 -f SGRBG8 /dev/video0  " in uos

Expected result:
successful
Actual result:
usb_dev_comp_req: Assertion

lishuangx, 2018-10-23:
[External_System_ID] 261104

Mingyuan18, 2018-11-21:
Can't reproduce this issue in this environment: ACRN-Hypervisor 2018w43.5.140000p_106; SOS Version: 25920

closed 2018-11-21 by Mingyuan18