projectacrn/acrn-hypervisor #1588: HV: build: configuration is inefficient and does not support flexible defconfig

eternalNight, 2018-10-24:
The current kconfig-based configuration system in the hypervisor has the following issues:

1. It takes several seconds for dependency checking, which is hardly acceptable.
1. config.h/config.mk may not remade right after changing .config by oldconfig.

In addition, the current implementation of defconfig is not flexible enough to support the following scenarios.

1. By 'make' a config with all default values is used, instead of one based on a defconfig.
1. It is difficult to maintain multiple defconfigs for different boards running the same firmware.

eternalNight, 2018-10-29:
[External_System_ID] 262780

junjiemao1, 2018-10-29:
[External_System_ID] 262780

Mingyuan18, 2018-12-24:
A symbol for specifying defconfig file is added in Kconfig. And the oldconfig tool now reads the specified defconfig as the base config when a new .config is to be created.
the bug can be closed.

closed 2018-12-24 by Mingyuan18