projectacrn/acrn-hypervisor #1595: doc: add comprehensive API documentation in code

eternalNight, 2018-10-25:
API documentation shall be maintained in the code and get updated whenever the documented code changes.

For the hypervisor, device model and tools we use doxygen-style comments. Here are  templates for functions and data structures.

```
/**
 * @brief Initialize environment for Trusty-OS on a VCPU.
 *
 * More info here.
 *
 * @param[in]    vcpu Pointer to VCPU data structure
 * @param[inout] param guest physical address. This gpa points to
 *                     struct trusty_boot_param
 *
 * @return 0 - on success.
 * @return -EIO - (description when this error can happen)
 * @return -EINVAL - (description …)
 *
 * @pre vcpu != NULL
 * @pre param != 0UL
 *
 * @post the return value is non-zero if param is ....
 * @post …
 *
 * @remark The api must be invoked with interrupt disabled.
 * @remark (Other usage constraints here)
 */
```

```
/**
 * @brief An mmio request.
 *
 * More info here.
 */
struct mmio_request {
        uint32_t direction;  /**< Direction of this request. */
        uint32_t reserved; /**< Reserved. */
        int64_t address; /**< gpa of the register to be accessed. */
 
        /**
         * @brief Width of the register to be accessed. 
         * 
         * More multiple-paragraph descriptions.
         */
        int64_t size; 

        int64_t value; /**< Single-line comments can be appended after the field. */
} __aligned(8);
```

For kernel modules such as VHM, the [kernel-doc](https://www.kernel.org/doc/html/latest/doc-guide/kernel-doc.html) format shall be followed.

junjiemao1, 2018-10-29:
[External_System_ID] 262833

gvancuts, 2018-11-20:
@eternalNight could we update the template to have a single invokation of `@return`, even for multiple values? Otherwise we get as many "return" sections in the HTML output which isn't very readable.

So something like:
```
 * @return
 *         - 0 - on success.
 *         - \p -EIO - (description when this error can happen)
 *         - \p -EINVAL - (description …)
```

junjiemao1, 2018-11-21:
It seems I confused ''@return'' and ''@retval''. For most of our cases ''@retval'' is a better way of specifying the return values in integer. As an example,

```
 * @retval 0       Successfully emulated by registered handlers.
 * @retval IOREQ_PENDING The I/O request is delivered to VHM.
 * @retval -EIO    The request spans multiple devices and cannot be emulated.
 * @retval -EINVAL \p io_req has an invalid type.
 * @retval Negative on other errors during emulation.
```

These adjacent ''@retval'' will be merged into a single paragraph in the generated html just like the adjacent ''@param''.

@gvancuts @dbkinder Would you please check if this looks good to you? If so, I can update the template here and propose patches to update the existing code in one series. Thanks.

junjiemao1, 2018-11-21:
And for functions with no return value, I would suggest we stick to ''@return None''. Will also update this in this issue as well as in the code.

gvancuts, 2018-11-21:
This looks good to me, thanks @junjiemao1 !

closed 2020-04-21 by fuzhongl