projectacrn/acrn-hypervisor #1602: ACRN console hang when it executed "reboot"

ftangx, 2018-10-25:
1.Environment
--------------------------------------------------------------
swupd info
Installed version: 25170
Version URL:       https://download.clearlinux.org/update/
Content URL:       https://cdn.download.clearlinux.org/update/
DM version is: 0.2-unstable-?”2018w38.4.140000p_91?” (daily tag:?”2018w38.4.140000p?”), build by mockbuild@2018-09-20 09:24:30
Linux clr-9637df086a924e95aa3be844702cc933 4.14.68-99.pk414-sos #1 SMP PREEMPT Fri Sep 14 22:03:41 UTC 2018 x86_64 GNU/Linux

2.  Steps
--------------------------------------------------------------
Hypervisor_VMM_Warm_Reboot　
1.In Hypervisor.
2."reboot'
3,Repeat steps 1-2 10 times

3. Current result:
--------------------------------------------------------------
step3: ACRN console hang

4. Expected result:
--------------------------------------------------------------
step3: system works well

5. repetition rate
100%

mwang106, 2018-10-26:
[External_System_ID] 256568

Mingyuan18, 2018-12-21:
based on this command "reboot" is just used for a debug function, and not as general function, it just can work under certain condition: after UOS running and SOS watchdog enabling, 
it can work. So we don't plan to resolve it, suggest to close this one.

closed 2018-12-21 by Mingyuan18