projectacrn/acrn-hypervisor #1605: acrn hypervisor part of latest clearlinux fails to boot guest OS

vijaydhanraj, 2018-10-25:
Launching guest VM crashes with the attached error. Based on the initial observation, the crash appears even before starting the guest kernel.

Clearlinux Version: 25800 (for both SOS and UOS)
Kernel version: 4.14.74-115

Please let me know if any other details are required.
[guest_launch_crash.txt](https://github.com/projectacrn/acrn-hypervisor/files/2516040/guest_launch_crash.txt)

gvancuts, 2018-10-29:
You need to reserve some hugepages first, as described in the error message. See this part:
```
to reserve pages (+orig 0): echo 2 > /sys/kernel/mm/hugepages/hugepages-1048576kB/nr_hugepages
now enough free pages are reserved!
```

Try doing that and **then** to launch the Guest OS.

Let us know if that helps!

mwang106, 2018-12-24:
@vijaydhanraj do you still have this problem?

vijaydhanraj, 2018-12-29:
No, the issue is resolved.

Thanks,
-Vijay

closed 2018-12-29 by mwang106