projectacrn/acrn-hypervisor #1639: [MRB][IO Mediator][SIT] USB_Mediator-SSD in uos, when write or read it, UOS appear Panic Reset

lidongwangx, 2018-10-29:
Environment:
HW: APL MRB
Build link:
MRB Daily test(Pre-merge#5B2):
SOS : http://10.239.147.147/acrn_daily_build/ww43.5/MRB/B2/
Image info:
acrn-hypervisor: acrn-2018w43.4-140000p
linux-pk414: Linux sos-dom0 4.19.0-rc8-quilt-2e5dc0ac-00001-gd8b14388695a #1 SMP PREEMPT Fri Oct 26 06:07:46 UTC 2018 x86_64 GNU/Linux
SOS version: 25800
linux-pk version: 4.19.0-rc8-quilt-2e5dc0ac-g284805005401
UOS OS Type: AaaG
Linux localhost Linux localhost 4.19.0-rc8-quilt-2e5dc0ac-g284805005401 #1 SMP PREEMPT Wed Oct 24 19:29:04 UTC 2018 x86_64

Reproduce rate: 100%

Reproduce steps:
-------------
1: one USB hub connect on usb3.0 port,hub plug with SSD devices

2:launch script
-s 7,xhci,1-1:1-2:1-3:2-1:2-2:2-3:cap=apl \
-s 8,passthru,0/15/1 \

3: uos can boot up successfully and can recogize all the Devices，total device nums: 8pcs;
console:/ # ls /dev/block/sd*
/dev/block/sda /dev/block/sda1 /dev/block/sda2

6:mount the device and read write it.
console:/ # mount dev/block/sda1 /data/usb2
console:/data/usb2 # 
console:/data/usb2 # dd if=/dev/zero of=test bs=1m count=5
[ 247.286720] Call Trace:
[ 247.288813] dump_stack+0x67/0x9b
[ 247.291846] panic+0xe3/0x233
[ 247.294369] ? _raw_spin_unlock_irqrestore+0x24/0x50
[ 247.298660] watchdog+0x263/0x3c0
[ 247.301701] ? reset_hung_task_detector+0x10/0x10
[ 247.306037] kthread+0x11e/0x140
[ 247.308876] ? kthread_create_worker_on_cpu+0x70/0x70
[ 247.313261] ret_from_fork+0x3a/0x50
[ 247.316639] reboot: panic mode set: p,w
[ 247.320025] Kernel Offset: 0x5000000 from 0xffffffff81000000 (relocation range: 0xffffffff80000000-0xffffffffbfffffff)
[ 247.329140] trusty-log panic notifier - trusty version Built: 19:08:54 Oct 24 2018
[ 247.335188] Rebooting in 10 seconds..

Expected result:
Step 6,uos can write and read the Devices

Actual result:
Step 6,operate timeout,appear Panic Reset.

lidongwangx, 2018-10-29:
[External_System_ID] ACRN-1612

Mingyuan18, 2018-11-28:
Can't reproduce this issue now: Hypervisor: acrn-2018w45.5-173000p;  SOS: 26200

closed 2018-11-28 by Mingyuan18