projectacrn/acrn-hypervisor #1657: UOS is assigned with 2G memory while 6G is expected.

dandanyx, 2018-10-30:
Environment:
------------
SOS kernel: 4.19.0-13.iot-lts2018-sos
UOS kernel: 4.19.0-rc8-quilt-2e5dc0ac-ged64d6cd2b9a / 9
Clearlinux Version: 25920
Hardware: ApolloLake
Reproduce rate: 100%

Bug detailed description:
-------------
UOS is assigned with 2G memory while 6G is expected
The MRB memory is 8G,but through launch_uos.sh get the MRB memory is slightly smaller than 8G

Reproduce steps:
-------------
1. Boot up SOS
2. Boot up UOS on "./launch_uos.sh -V 2" in SOS
3. Run command "free -h" on UOS get the uos memory
gordon_peak_acrn:/ $ free -h
                total        used        free      shared     buffers
Mem:             1.9G        1.8G         95M        164M         21M
-/+ buffers/cache:           1.8G        116M
Swap:               0           0           0

Current result:
--------------
UOS is assigned with 2G memory

Expected result:
--------------
UOS is assigned with 6G memory

Basic root-causing log:
--------------
no log

dandanyx, 2018-10-30:
[External_System_ID] 263056

Mingyuan18, 2018-11-28:
Can't reproduce this issue now: ACRN-Hypervisor 2018w44.4.140000p_112; SOS version: 26030

closed 2018-11-28 by Mingyuan18