projectacrn/acrn-hypervisor #1670: [IO Mediator][SIT] touch with USB3.0 port in uos, when touch the screen , it always print Error Infor: not enough avail descs

lidongwangx, 2018-10-31:
Environment Image info:
acrn-hypervisor: acrn-2018w44.1-140000p
linux-pk414: Linux sos-dom0 4.19.0-rc8-quilt-2e5dc0ac-00010-g689bffb94893 #1 SMP PREEMPT Tue Oct 30 01:30:38 UTC 2018 x86_64 GNU/Linux
SOS version: 25920
linux-pk version: 4.19.0-rc8-quilt-2e5dc0ac-g284805005401
UOS OS Type: AaaG
Linux localhost 4.19.0-rc8-quilt-2e5dc0ac-g284805005401 #1 SMP PREEMPT Wed Oct 24 19:29:04 UTC 2018 x86_64
Reproduce rate: 100%

Reproduce steps:
-------------
1: insert touch label into usb3.0 port or usb2.0 port.

2. check input device info. And find the input event in SOS:
root@sos-dom0 /home # cat /proc/bus/input/devices
I: Bus=0003 Vendor=04f3 Product=24e1 Version=0110
N: Name="ELAN Touchscreen"
P: Phys=usb-0000:00:15.0-2/input0
S: Sysfs=/devices/pci0000:00/0000:00:15.0/usb1/1-2/1-2:1.0/0003:04F3:24E1.0004/input/input11
U: Uniq=
H: Handlers=mouse0 event1
B: PROP=2
B: EV=1b
B: KEY=400 0 0 0 0 0
B: ABS=3273800000000003
B: MSC=20

4. add virtio-input parameter in launch.sh:
-s 7,virtio-input,/dev/input/event1
5. boot up uos and check touch take effect on uos
when touch the screen, it always print Error Infor:console:/ $virtio_input_send_event: not enough avail descs, dropped:69

Expected result:
Step 5,when touch the screen, it should not print related error information.

Actual result:
Step 5,when touch the screen, it always print Error Infor: console:/ $virtio_input_send_event: not enough avail descs, dropped:69

Debugging info:
console:/ $virtio_input_send_event: not enough avail descs, dropped:69

Attachement:
no need the attachment. console:/ $virtio_input_send_event: not enough avail descs, dropped:69

lidongwangx, 2018-10-31:
[External_System_ID] ACRN-1619

Mingyuan18, 2018-11-28:
Can't reproduce this issue now: Hypervisor: 2018w44.4.140000p; SOS: 26030

closed 2018-11-28 by Mingyuan18