projectacrn/acrn-hypervisor #1727: [MRB][IO Mediator][AaaG] UOS(AaaG) resume from S3 failed

lishuangx, 2018-11-06:
Environment:
HW: APL MRB

Build link:
https://mcg-depot.intel.com/artifactory/cactus-absp-jf/master-release_candidate/940/gordon_peak_acrn/userdebug/

General image Information:
>Kernel Version/Android Version	4.19.0-quilt-2e5dc0ac-gab297fcae002 / 9
AOSP Version	PPR1.180610.011
ABL Version	1844_GP20
IFWI Version	3.1.55.2278
IOC Version	4.0.11
ACRN-Hypervisor 2018w44.4.140000p_112
SOS Build Link	20181102-26030
SOS Version	Installed version: 26030
SOS Kernel Version	4.19.0-15.iot-lts2018-sos #1 SMP PREEMPT Fri Nov 2
cAVS FW Version	9.22.01.3774

Reproduce rate: 100%

Reproduce steps:
1. launch sos 
2. add vm via "acrnctl add ./launch_uos.sh -V 2" in sos
3. run "acrnd & " in sos
4. after uos boot up , press ignition key to make sos&uos suspend
5.After sos&uos enter sleep mode
6.Press ignition key to resume  sos&uos

EXPECTED OUTCOME
==========================================================
After step 6,SOS & UOS can resume normally

ACTUAL OUTCOME
==========================================================
After step6, sos can resume but uos resume failed

lishuangx, 2018-11-06:
[External_System_ID] ACRN-1649

Mingyuan18, 2018-12-19:
Can't reproduce this issue in below build: 
Hypervisor: 2018w45.5.140000p_116
SOS : 26200

closed 2018-12-19 by Mingyuan18