projectacrn/acrn-hypervisor #1773: USB Mediator: Can't find all devices when multiple usb devices connected[Reproduce rate:60%]

mwang106, 2018-11-09:
Environment:
HW: APL NUC

Build link:
Please refer the E2E NUC image
https://download.clearlinux.org/latest (26120) Which tag is 2018w45.1.140000p

Image info:
SOS version: 26120
SOS kernel version: Linux robin-nuc 4.19.0-16.iot-lts2018-sos #1 SMP PREEMPT Mon Nov 5 20:36:30 UTC 2018 x86_64 GNU/Linux
UOS Type: LaaG
UOS version: 26120
UOS kernel version: Linux clr-ba4d04dad3654e2f9075550351e68c35 4.19.0-16.iot-lts2018 #1 SMP Mon Nov 5 20:35:04 UTC 2018 x86_64 GNU/Linux

Reproduce rate: 60%

Reproduce steps:
-------------
1. insert mutiple devices(SSD、USB2.0 and USB3.0 flash disk) in standard A port.

2:boot uos by sharing the USB stick in cmd line
-s n,xhci,1-1:1-2:1-3:2-1:2-2:2-3 \

3:uos can access USB mass storage.
root@clr-ba4d04dad3654e2f9075550351e68c35 ~ # fdisk -l |grep dev
Disk /dev/vda: 8.5 GiB, 9161408512 bytes, 17893376 sectors
/dev/vda1 2048 1048575 1046528 511M EFI System
/dev/vda2 1048576 1114111 65536 32M Linux swap
/dev/vda3 1114112 17891327 16777216 8G Linux root (x86-64)
Disk /dev/sda: 14.3 GiB, 15376318464 bytes, 30031872 sectors — (USB2.0 flash disk)
/dev/sda1 2050048 30031871 27981824 13.4G 83 Linux
/dev/sda2 2048 2048000 2045953 999M 83 Linux
Disk /dev/sdb: 58.1 GiB, 62411243520 bytes, 121896960 sectors — (USB3.0 flash disk)
/dev/sdb1 2050048 121896959 119846912 57.2G 83 Linux
/dev/sdb2 2048 2048000 2045953 999M 83 Linux
Disk /dev/sdc: 111.8 GiB, 120034123776 bytes, 234441648 sectors — (SSD disk)
/dev/sdc1 2097120 234441647 232344528 110.8G 83 Linux
/dev/sdc2 65535 2048000 1982466 968M 83 Linux

4: hot plug USB mass storage in UOS

5: uos can access USB mass storage，not find the USB3.0 flash disk；
oot@clr-ba4d04dad3654e2f9075550351e68c35 ~ # fdisk -l |grep dev
Disk /dev/vda: 8.5 GiB, 9161408512 bytes, 17893376 sectors
/dev/vda1 2048 1048575 1046528 511M EFI System
/dev/vda2 1048576 1114111 65536 32M Linux swap
/dev/vda3 1114112 17891327 16777216 8G Linux root (x86-64)
Disk /dev/sdc: 111.8 GiB, 120034123776 bytes, 234441648 sectors
/dev/sdc1 2097120 234441647 232344528 110.8G 83 Linux
/dev/sdc2 65535 2048000 1982466 968M 83 Linux
Disk /dev/sda: 14.3 GiB, 15376318464 bytes, 30031872 sectors
/dev/sda1 2050048 30031871 27981824 13.4G 83 Linux
/dev/sda2 2048 2048000 2045953 999M 83 Linux

Expected result:
Step 5,Can recognize the USB3.0 flash disk in the UOS.

Actual result:
Step 5,Can not recognize the USB3.0 flash disk in the UOS.

Debugging info:
please see the attachment.

Attachement:
please see the attachment.

mwang106, 2018-11-09:
[External_System_ID] ACRN-1659

fuzhongl, 2020-04-27:
Fixed with enlarge interval time between plug and unplug.

closed 2020-04-27 by fuzhongl