projectacrn/acrn-hypervisor #1776: [APLUP2]ACRN debugging tool - acrnlog cannot be used in SOS

mwang106, 2018-11-09:
Environment
[Board]: APL UP2

Installed version: 26120
Version URL: https://cdn.download.clearlinux.org/update/
Content URL: https://cdn.download.clearlinux.org/update/

2. Steps
1). Launch SOS
2). Check: ls /dev/acrn_hvlog*

3. Current result:
acrn_hvlog is not exist in SOS.

4. Expected result:
acrn_hvlog is exist in SOS.

5. repetition rate
100%

6. logs
root@sos-dom0 ~ # ls /dev/
acrn_vhm full i2c-12 loop3 mqueue ram14 sg0 tty17 tty31 tty46 tty60 usbmon2 vcsa5
ashmem fuse i2c-2 loop4 net ram15 shm tty18 tty32 tty47 tty61 vbs_hyper_dmabuf vcsa6
binder gpiochip0 i2c-3 loop5 network_latency ram2 snd tty19 tty33 tty48 tty62 vbs_ipu vcsu
block gpiochip1 i2c-4 loop6 network_throughput ram3 stderr tty2 tty34 tty49 tty63 vbs_rng vcsu1
bsg gpiochip2 i2c-5 loop7 null ram4 stdin tty20 tty35 tty5 tty7 vcs vcsu2
bus gpiochip3 i2c-6 loop-control nvram ram5 stdout tty21 tty36 tty50 tty8 vcs1 vcsu3
char gpt-auto-root i2c-7 mapper port ram6 tty tty22 tty37 tty51 tty9 vcs2 vcsu4
console hugepages i2c-8 mei0 ppp ram7 tty0 tty23 tty38 tty52 ttyS0 vcs3 vcsu5
core hv_npk_log i2c-9 memory_bandwidth ptmx ram8 tty1 tty24 tty39 tty53 ttyS1 vcs4 vcsu6
cpu hwbinder initctl mmcblk0 pts ram9 tty10 tty25 tty4 tty54 ttyS2 vcs5 vga_arbiter
cpu_dma_latency hwrng input mmcblk0boot0 ram0 random tty11 tty26 tty40 tty55 ttyS3 vcs6 vndbinder
disk hyper_dmabuf kmsg mmcblk0boot1 ram1 rfkill tty12 tty27 tty41 tty56 uhid vcsa zero
dri i2c-0 log mmcblk0p1 ram10 rtc tty13 tty28 tty42 tty57 uinput vcsa1 zram0
efi_capsule_loader i2c-1 loop0 mmcblk0p2 ram11 rtc0 tty14 tty29 tty43 tty58 urandom vcsa2
fb0 i2c-10 loop1 mmcblk0p3 ram12 sda tty15 tty3 tty44 tty59 usbmon0 vcsa3
fd i2c-11 loop2 mmcblk0rpmb ram13 sda1 tty16 tty30 tty45 tty6 usbmon1 vcsa4

mwang106, 2018-11-09:
[External_System_ID] ACRN-1676

Mingyuan18, 2019-02-13:
the bug cannot reproduced with 
Installed version: 27600
Version URL: http://linux-ftp.sh.intel.com/pub/mirrors/clearlinux/update
Content URL: http://linux-ftp.sh.intel.com/pub/mirrors/clearlinux/update

closed 2019-02-13 by Mingyuan18