projectacrn/acrn-hypervisor #1777: After UOS played video for several minutes, the UOS image was stagnant

mwang106, 2018-11-09:
Environment:
------------
NUC image: 
Installed version: 26120
Version URL: https://cdn.download.clearlinux.org/update/
Content URL: https://cdn.download.clearlinux.org/update/

Bug detailed description:
-------------
After UOS played video for several minutes, the UOS image was stagnant.

Reproduce steps:
-------------
1. Boot up SOS& UOS
2. UOS execute：
systemctl enable ias
systemctl start ias
export XDG_RUNTIME_DIR=/run/ias 
systemctl enable ias
systemctl start ias
export XDG_RUNTIME_DIR=/run/ias

3.UOS execute:
st-launch-1.0 -v filesrc location=ToS_1080p_59.94fps_H264_25000kbps_8bits_noHDR_barcoded-v1511090000.mp4 ! qtdemux ! vaapidecodebin ! waylandsink

Current result:
--------------
After UOS played video for several minutes, the UOS image was stagnant.

Expected result:
--------------
UOS play video image smoothing all the time

Failed log:
--------------
no log

mwang106, 2018-11-09:
[External_System_ID] ACRN-1674

Mingyuan18, 2018-11-28:
Can't reproduce this issue in SOS:26200.

closed 2018-11-28 by Mingyuan18