projectacrn/acrn-hypervisor #1778: MSDK failed to decode H264 1920*1080 with HW

mwang106, 2018-11-09:
Environment:
------------
E2E NUC image: 
https://download.clearlinux.org/latest (26120) Which tag is 2018w45.1.140000p

Bug detailed description:
-------------
MSDK case "DOMU: MSDK decode H264 1920*1080 with HW" test failed in UOS

Reproduce steps:
-------------
1. Boot up SOS& UOS
2. install MSDK in UOS:
Copy the whole folder (http://10.239.143.118/Clear_Linux_Test_Tools/msdk_install/) to SOS or LaaG
Run the command "./installMSDK.sh" and check the output (should contain below info)
sample_decode succeeds.
sample_encode succeeds.
3. export LIBVA_DRIVER_NAME=iHD
chmod +x /usr/lib64/dri/iHD_drv_video.so
cd /usr/share/mfx/samples
4. Executed below command:
./sample_decode h264 -hw -i test_stream_1920x1080.264 -o test_stream.yuv

Current result:
--------------
Test failed

Expected result:
--------------
Test pass

Failed log:
root@clr-fa4eba32bbfd4d08ac822f92ada40df7 /usr/share/mfx/samples # ./sample_decode h264 -hw -i test_stream_1920x1080.264 -o test_stream.yuv
libva info: VA-API version 1.3.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/lib64/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_2
libva info: va_openDriver() returns 0
Decoding Sample Version 8.3.26.

Input video AVC
Output format NV12
Input:
Resolution 1920x1088
Crop X,Y,W,H 0,0,1920,1080
Output:
Resolution 1920x1080
Frame rate 29.97
Memory type system
MediaSDK impl hw
MediaSDK version 1.27

Decoding started
Frame number: 186, fps: 20.524, fread_fps: 0.000, fwrite_fps: 49.666
[ERROR], sts=MFX_ERR_UNDEFINED_BEHAVIOR(-16), WriteNextFrame, pInfo.CropW!=fwrite(pData.UV + (pInfo.CropY * pData.Pitch / 2 + pInfo.CropX / 2) + i * pData.Pitch, 1, pInfo.CropW, dstFile) error at /home/zhanqi/workspace_ronnie/MSDK/msdk/samples/sample_common/src/sample_utils.cpp:876
Frame number: 186, fps: 20.331, fread_fps: 0.000, fwrite_fps: 48.632
[ERROR], sts=MFX_ERR_UNKNOWN(-1), RunDecoding, Unexpected error!! at /home/zhanqi/workspace_ronnie/MSDK/msdk/samples/sample_decode/src/pipeline_decode.cpp:1939

[ERROR], sts=MFX_ERR_UNKNOWN(-1), main, Pipeline.RunDecoding failed at /home/zhanqi/workspace_ronnie/MSDK/msdk/samples/sample_decode/src/sample_decode.cpp:705
root@clr-fa4eba32bbfd4d08ac822f92ada40df7 /usr/share/mfx/samples #

mwang106, 2018-11-09:
[External_System_ID] ACRN-1665

Mingyuan18, 2018-11-28:
Can't reproduce this issue in : Acrn-hypervisor: 2018w45.5.140000p; SOS: 26200

closed 2018-11-28 by Mingyuan18