projectacrn/acrn-hypervisor #1780: Some video formats cannot be played in SOS.

mwang106, 2018-11-09:
Environment:
------------
E2E NUC image: 
https://download.clearlinux.org/latest (26120) Which tag is 2018w45.1.140000p

Bug detailed description:
-------------
MSDK case "DOMU: MSDK decode H264 1920*1080 with HW" test failed in UOS

Bug detailed description:
-------------
The following format of video cannot be played on SOS.
1.H265_10bits 
2.VP8
3.VP9
4.VP9_10bits
5.H265.720p

Reproduce steps:
-------------
1. Boot up hypervisor and SOS and UOS
2. do settings in SOS:
systemctl enable ias
systemctl start ias
export XDG_RUNTIME_DIR=/run/ias
3.mount /dev/sda1 /mnt
4.gst-launch-1.0 -v filesrc location=/mnt/Performance_Video_Clips/H265_10bits/ToS_2160p_59.94fps_H265_35000kbps_10bits_noHDR_barcoded-v1511090000.mp4 ! qtdemux ! vaapidecodebin ! waylandsink

Current result:
--------------
There are some formats of video that cannot be played on SOS.

Expected result:
--------------
H265_10bits,VP8,VP9,VP9_10bits can be played on SOS

Attachments

mwang106, 2018-11-09:
[External_System_ID] ACRN-1671

Mingyuan18, 2019-01-29:
Close this issue due to it is third party issue.

Mingyuan18, 2019-03-07:
This issue can't reproduce in below build:
SOS: 28100
ACRN Tag: 2019w09.5.140000p

closed 2019-03-07 by Mingyuan18