projectacrn/acrn-hypervisor #1781: Can not recognize the SD card in the SOS

mwang106, 2018-11-09:
Environment:
HW: APL NUC

Image info:
SOS version: 26050
SOS kernel version: Linux clr-c62bc84245f0442aad70d0e9ad779ad6 4.19.0-15.iot-lts2018-sos #1 SMP PREEMPT Thu Nov 1 10:35:20 UTC 2018 x86_64 GNU/Linux
UOS Type: LaaG
UOS version: 26050
UOS kernel version: Linux clr-3bb03b65df7646dbabc5f1728675bf78 4.19.0-15.iot-lts2018 #1 SMP Thu Nov 1 10:33:54 UTC 2018 x86_64 GNU/Linux

Reproduce rate: 100%

Reproduce steps:
-------------
1. insert SD card into NUC.

2. Can not recognize the SD card in the SOS.

Expected result:
Step 2,Can recognize the SD card in the SOS..

Actual result:
Step 3,Can not recognize the SD card in the SOS.

Debugging info:
please see the attachment.

Attachement:
please see the attachment.

mwang106, 2018-11-09:
[External_System_ID] ACRN-1646

Mingyuan18, 2018-11-28:
Can't reproduce this issue in below environment: Hypervisor: 2018w45.5.140000p; SOS:26200

closed 2018-11-28 by Mingyuan18