projectacrn/acrn-hypervisor #1782: UOS failed to get ip with the pass-throughed network card

mwang106, 2018-11-09:
1: enable VT-d and boot sos; 
2:hide network card in sos
1): find out network card
$lspci 
2):Find out the numeric PCI ID for network card
$lspci -n
02:00.0 0200: 8086:1533 (rev 03)
3):edit parameters：
echo "8086 1533" > /sys/bus/pci/drivers/pci-stub/new_id
echo "0000:02:00.0" > /sys/bus/pci/devices/0000:02:00.0/driver/unbind
echo "0000:02:00.0" > /sys/bus/pci/drivers/pci-stub/bind

-s n,passthru,02/00/0
3:boot uos by pass-through this network card in cmd line.
4:uos can get IP and ping is ok, and sos cannot find this network card.

mwang106, 2018-11-09:
[External_System_ID] ACRN-1679

Mingyuan18, 2019-02-19:
Cannot reproduce the issue
build info:
Installed version: 27840
Version URL: https://cdn.download.clearlinux.org/update/
Content URL: https://cdn.download.clearlinux.org/update/
ACRN tag:”2019w07.5.140000p”

closed 2019-02-19 by Mingyuan18