projectacrn/acrn-hypervisor #1797: Failed to execute the command "./launch_uos.sh -V 2" to boot up UOS in AaaG Create/Destory stability test.

dandanyx, 2018-11-12:
Environment:
------------
SOS version: 20181029-25920
SOS Kernel Version: 4.19.0
acrn-hypervisor: 2018w43.5.140000p_106
IOC Version:4.0.8
ABL Version:rel.1836

Bug detailed description:
-------------
Failed to execute the command "./launch_uos.sh -V 2" to boot up UOS.

Log info:
+ ssh 10.239.81.162 -C './launch_uos.sh -V 2'
cat: /sys/class/mei/mei0/hbm_ver: No such file or directory
./launch_uos.sh: line 391:   781 Segmentation fault      (core dumped) mount /dev/mmcblk1p3 /data
cpu1 online=0
cpu2 online=0
cpu3 online=0
Launch android UOS
no /data/android/android.img, exit

Reproduce steps:
-------------
1. Boot up hypervisor and SOS
2. In SOS, execute the command "./launch_uos.sh -V 2" to boot up UOS.
3. After UOS boot up, execute the command "adb reboot -p" to shutdown UOS.
4. Return to step 2, repeatedly running AaaG Create/Destory stability test.

Current result:
--------------
Failed to execute the command "./launch_uos.sh -V 2" to boot up UOS.
Failed rate: 1/10

Expected result:
--------------
Can succeed to execute the command "./launch_uos.sh -V 2" to boot up UOS.

Attachment
--------------
no log

dandanyx, 2018-11-12:
[External_System_ID] 263554

Mingyuan18, 2018-12-11:
Can't reproduce this issue in this build: Hypervisor: 2018w45.5.140000p_116; SOS: 20181112-26200

closed 2018-12-11 by Mingyuan18