projectacrn/acrn-hypervisor #1821: Optimize IO request path

shuox, 2018-11-15:
Now, the sketch of ioreq process can be,
  A. UOS vcpu0 generate PIO/MMIO ->
   B. pcpu1(vcpu0 of UOS) trap into HV ->
    C. pcpu1 build ioreq, send IPI and enter idle ->
     D.1 pcpu0(vcpu0 of SOS) response IPI,
     D.2 pcpu0 handle the ioreq in HV, kernel, DM,
     D.3 pcpu0 mark ioreq as complete,
     D.4 pcpu0 hypercall to enter HV ->
       E.1 pcpu0 send IPI to wake pcpu1 up
       E.2 UOS vcpu0 continue running

we can skip D.4, E.1 steps. In step C, pcpu1 will enter a
polling ioreq state idle after send out the IPI.
It can save about ~5000 cpu cycles.

We can do the polling in idle instead of pause cpu all the
time. It will consume more power. A better way is to use monitor/mwait
instructions which can put cpu into a sleep state with monitoring a
memory address. Unfortunately, APL has bug with monitor. We can gather
all ioreqs state into one monitorable memory and take advantage of
monitor/mwait for future platform.

I did some performance tests as comparison,
1) 
  virtio-net, signle thread UDP, UOS->PC:
  server: iperf -s -u -V
  client: iperf -c <IP> -u -b 1G -t 600 -V
  
  w/o  patch, Bandwidth: 353.6 Mbits/sec
  with patch, Bandwidth: 383.3 Mbits/sec
  improve 8.5%

2)
  USB mediator disk read/write:
  command: iozone -ec -+n -I -L64 -S32 -r4K -s100m -i0 -i1 -f iozone.tmp
  w/o  patch,  write: 1603, read: 3148
  with patch,  write: 1609, read: 3199
  improve             0.37%       1.62%

3)
  vitioblk iozone:
  command: iozone -ec -+n -I -L64 -S32 -r4K -s100m -i0 -i1 -i2 -f iozone.tmp
               write    read     random write    random read
  w/o  patch   786.75  61078.125   59923.25        786.625
  with patch   848.63  63469.72    62526.72        842
   improve     7.88%    3.92%       4.34%          7.04%

4)
  acrngt on kernel 4.14 (ClearLinux as Guest):
  command: glmark2-es2-wayland
  w/o  patch, 883
  with patch, 902
   improve    2.2%

shuox, 2018-11-15:
[External_System_ID] ACRN-1722

NanlinXie, 2019-01-10:
Closed with no regression happen.

closed 2019-01-10 by NanlinXie