projectacrn/acrn-hypervisor #1831: Documentation now broken regarding starting Weston

mattcurfman, 2018-11-15:
Section 7 of the documentation at https://projectacrn.github.io/latest/getting-started/apl-nuc.html#software-setup shows how to start Weston in the SOS and Guest OS.  Apparently these steps are now broken by inclusion of IAS version of weston.  Can the documentation be updated to show the correct method to start IAS Weston in both SOS and Guest OS?

This error is now displayed when following the steps documented:
```
Failed to load module: /usr/lib64/weston/ias-shell.so: cannot open shared object file: No such file or directory
```

mattcurfman, 2018-11-15:
Kevron mentions the correct way to instantiate IAS Weston is now:
```
# systemctl start ias
```

There is also some question how this relates to the user created (ias requires a special user?) so this should be discussed with him and verified.

mattcurfman, 2018-11-15:
Using 'systemctl start ias', I now see the following error:
```
Nov 15 22:41:46 clr-6b0c3d3d04c2485ab1bf8193d7e14748 ias-weston-launch[1028]: [22:41:46.812] IAS ERROR: Could not build any suitable CRTC's
Nov 15 22:41:46 clr-6b0c3d3d04c2485ab1bf8193d7e14748 ias-weston-launch[1028]: [22:41:46.812] failed to create crtcs for /sys/devices/pci0000:00/0000:00:02.0/drm/card0
Nov 15 22:41:46 clr-6b0c3d3d04c2485ab1bf8193d7e14748 ias-weston-launch[1028]: [22:41:46.814] fatal: failed to create compositor backend
```

hwang37, 2018-11-16:
@mwang106  could you link to our known issue and set Tracked-On ID?

mwang106, 2018-11-16:
Linked with #1794

mwang106, 2019-01-16:
#1794 closed

closed 2019-01-16 by mwang106