projectacrn/acrn-hypervisor #1837: 'acrnctl list' shows incomplete VM names

liuming50, 2018-11-16:
I am currently using Yocto build and deploy UOS images, it would be much more convenient to let the VMs to have the same names with when they are being built. For instance, a typical image name in Yocto is like: uos-image-weston, but it can not show the name completely with 'acrnctl list', since the length limit is 16.

Increasing the limit to 32 would fix this problem.

mwang106, 2018-11-19:
[External_System_ID] ACRN-1744

lijinxia, 2018-11-19:
[External_System_ID] ACRN-1746

ftangx, 2018-11-21:
The bug cannot be reproduced in NUC and UP2.
Installed version: 26410
Version URL: http://linux-ftp.sh.intel.com/pub/mirrors/clearlinux/update
Content URL: http://linux-ftp.sh.intel.com/pub/mirrors/clearlinux/update

liuming50, 2018-11-21:
It's reproducible with the build from https://github.com/liuming50/meta-acrn-manifest, the patch had been merged so close it.

closed 2018-11-21 by liuming50