projectacrn/acrn-hypervisor #1844: Establish a `@return` keyword for functions not returning any value (`void function()`)

gvancuts, 2018-11-19:
We should establish a "preferred" method to indicate in Doxygen that a function does not return any value (`void function()`).

This is inconsistent throughout the ACRN codebase at the moment, we have instances of these being used for functions not returning anything:
```
@return void
@return N/A
@return None
@return NULL
```

_Originally posted by @gvancuts in https://github.com/projectacrn/acrn-hypervisor/pull/1820/comment#issuecomment-439859846_

Mingyuan18, 2019-03-22:
@gvancuts , Does this issue still need to keep open?

gvancuts, 2019-03-22:
@junjiemao1 provided a couple of PRs that updated the current codebase - see:
584f6b72 doc: replace return with retval
97eb72a4 doc: always use 'None' for functions not returning a value

Having said that, my intention with this feature request was also to more formally establish guidelines so developers and contributors know which we prefer in ACRN. Unless I have missed this, there is no such recommendation in our guidelines documentation.

junjiemao1, 2019-03-25:
@shiqingg is helping drafting the design and coding guidelines.

@shiqingg Would you please check if the format of doxygen-style comments are properly recorded in any of our guidelines? If not, we can add the templates and rules in our coding guideline [here](https://projectacrn.github.io/latest/developer-guides/coding_guidelines.html#).

shiqingg, 2019-03-25:
> @shiqingg is helping drafting the design and coding guidelines.
>
> @shiqingg Would you please check if the format of doxygen-style comments are properly recorded in any of our guidelines? If not, we [...]

We don't have rules for doxygen-style comments currently.
I will add some rules in our coding guidelines.

Mingyuan18, 2019-04-04:
[External_System_ID] ACRN-3299

Mingyuan18, 2019-05-06:
@gvancuts,  May I ask whether this issue can be closed now?

gvancuts, 2019-05-06:
Yes, closing!

closed 2019-05-06 by gvancuts