projectacrn/acrn-hypervisor #1846: Aaag guest hang when execute "adb reboot

tomasbw, 2018-11-19:
CaseName: Hypervisor_Bootloader_Aaag_vsbl_003
Reproduce step:
1. Launch SOS 
2. Use acrn-dm to launch android with -vsdl <valiad vsbl image path >
(default ./launch_UOS.sh or ./launch_UOS.sh -V 2)
3. Execute "adb reboot" more times

3. Expected result
--------------------------------------------------------------
UOS reboot sucessfully

4. Current result
--------------------------------------------------------------
step4: When we run step4 more times, Aaag guest will hang at the second times.

5. repetition rate
1/10

tomasbw, 2018-11-19:
[External_System_ID] ACRN-1706, ACRN-1705

Mingyuan18, 2018-12-14:
Can't reproduce this issue in below build: ACRN:  2018w48.1.173000p_123; SOS : 20181128-26540

closed 2018-12-14 by Mingyuan18