projectacrn/acrn-hypervisor #1857: SOS kernel panic & restart due to "BUG: spinlock bad magic on CPU#0"

mwang106, 2018-11-20:
Environment:
------------
Kernel Version/Android Version: 4.19.0-quilt-2e5dc0ac-g088074fda9fe / 9
AOSP Version: PPR1.180610.011
ABL Version: 1844_GP20
IFWI Version: 3.1.55.2278a
IOC Version: 4.0.12
ACRN-Hypervisor: 2018w45.5.140000p_116
SOS Build Link: 20181112-26200
SOS Version Installed version: 25660
SOS Kernel Version: 4.19.0-19.iot-lts2018-sos #1 SMP PREEMPT Mon Nov 12
cAVS FW Version: 9.22.01.3774

Bug detailed description:
-------------
In the stability test "AaaG Reboot", SOS kernel panic and system reboot after Android guest boot up.
When this issue occurred, AaaG reboot test ran 160 cycles and this test has ran for 4 hours.

Reproduce steps:
-------------
1. Boot up hypervisor and SOS
2. Execute "./launch_uos.sh -V 2" to boot up Android guest (removed audio/IPU/CSE mediator lines in launch_uos.sh script )
3. After wait for 60s until Android guest boot up successfully
4. Execute "adb shell reboot" to reboot Android guest
5. Return to step 3

Current result:
--------------
In the stability test "AaaG Reboot", SOS kernel panic and system reboot after Android guest boot up.
When this issue occurred, AaaG reboot test ran 160 cycles and this test has ran for 4 hours.

Expected result:
--------------
No kernel panic, no system reboot.

Attachment
--------------
console log

mwang106, 2018-11-20:
[External_System_ID] ACRN-1719

Mingyuan18, 2018-12-19:
This issue is to track spinlock issue caused by race issue between ioreq client destroy. Not reproduced in below build, therefore close it. 
ACRN-Hypervisor: 2018w48.1.173000p_123; SOS: 26540

closed 2018-12-19 by Mingyuan18