projectacrn/acrn-hypervisor #1860: [Fuzzing]Hypervisor carsh:Assertion failed in file common/ptdev.c,line 135 : fatal error

mwang106, 2018-11-20:
We used Syzkaller ran with hypercall unit tests to do Fuzzing test for ACRN, hit hypervisor crash:

Hypervisor crash logs as below：

[9367321us][cpu=0][sev=3][seq=42]:VM 1 GPA2HPA: failed for gpa 0xfee00000
[9374688us][cpu=0][sev=3][seq=43]:modify_or_del_pdpte, invalid op, pdpte not present

[11162603us][cpu=1][sev=3][seq=52]:vlapic: Start Secondary VCPU1 for VM[1]...
[11178656us][cpu=1][sev=3][seq=54]:vlapic: Start Secondary VCPU2 for VM[1]...
[57448227us][cpu=0][sev=3][seq=56]:[request_irq] invalid irq num
Assertion failed in file common/ptdev.c,line 135 : fatal error
Host Stack: CPU_ID = 0
addr(0x6e04abe8)        0x000000006f041b60  0x000000000000fbff  0x000000000077fbff  0x000000006e01bb30
addr(0x6e04ac08)        0x0000000000000004  0xffffffea00770004  0x0000000000770004  0x000000006e00ab80
addr(0x6e04ac28)        0x0000000000000020  0x000000000000fbff  0x0000000040000000  0x0000000000000077
addr(0x6e04ac48)        0x000000000000fbff  0x0000000000000004  0x0000000000000100  0x000000006e066000
addr(0x6e04ac68)        0x0000000000000020  0x000000006e00b866  0x000000006e066000  0x000000006e16a000
addr(0x6e04ac88)        0x000000025b1d7de0  0x0000000000000001  0x000000025b1d7de0  0x000000006e01a7e1
addr(0x6e04aca8)        0x0004fbff00000001  0x0000040100000100  0x0000000000000000  0x9e7bd03a23c588dd
addr(0x6e04acc8)        0x000000006f03b2e0  0x0000000080000053  0x000000006e16a000  0x000000006e012e1f
addr(0x6e04ace8)        0x000000006e029235  0x0000000000000012  0x000000006f03b2e0  0x0000000000000012
addr(0x6e04ad08)        0x0000000000000020  0x9e7bd03a23c588dd  0x000000006f03b2e0  0x000000192aeb9372
addr(0x6e04ad28)        0x00000000c0000103  0x0000000000000020  0x0000000000000000  0x000000006e019192
addr(0x6e04ad48)        0x0000000000000000  0x000000006f03bc30  0x0000000000000000  0x000000192ae8047a
addr(0x6e04ad68)        0x0002000000000011  0x0000000000000000  0x0000000000000000  0x9e7bd03a23c588dd
addr(0x6e04ad88)        0x0000000000000000  0x000000006f03b2e0  0x0000000000000000  0x0000000000000000
addr(0x6e04ada8)        0x000000006e042c88  0x0000000000000000  0x0000000000000000  0x323238343437355b
addr(0x6e04adc8)        0x7570635b5d737537  0x3d7665735b5d303d  0x353d7165735b5d33  0x757165725b3a5d36

Host Call Trace:

================================================================================

Guest Registers:
=       VM ID 0 ==== vCPU ID 0 ===  pCPU ID 0 ====world 0 =============
=       RIP=0xffffffff81bb2559  RSP=0xffff88025b1d7da8 RFLAGS=0x0000000000000297
=       CR0=0x0000000080050033  CR2=0x000000c42037bb80  CR3=0x0000000266739000
=       RAX=0xffff88025b214000  RBX=0x0000000000000001  RCX=0x0000000000000000
=       RDX=0x0000000000000000  RDI=0x0000000000000001  RSI=0x000000025b1d7de0
=       RBP=0x000000025b1d7de0  R8=0x0000000080000053  R9=0x000000002aeb6b04
=       R10=0xffff88025b1d7ed0  R11=0x000000004406f74b  R12=0xffff880268905000
=       R13=0x000077ff80000000  R14=0x0000000000000001  R15=0x00007f7ab6ffcd38

Guest Stack:
Dump stack for vcpu 0, from gva 0xffff88025b1d7da8
guest_rsp(0xffff88025b1d7da8):  0xffff8802db1d7de0  0xffff88025b1d7e70  0xffffffff8168c028  0x0000000000000000
guest_rsp(0xffff88025b1d7dc8):  0x0004fbff00000001  0x0000040100000100  0x0000000000000000  0x0004fbff00000001
guest_rsp(0xffff88025b1d7de8):  0x0000040100000100  0x0000000000000000  0x0000000000000001  0x0000000043000053
guest_rsp(0xffff88025b1d7e08):  0xffff88025b1d7eb0  0x0000000000000001  0xffff880270ebb880  0x0000000000000001
guest_rsp(0xffff88025b1d7e28):  0x0000000043000000  0xffffffff8144db91  0xffffffff8144da90  0xffff880271ce7aa0
guest_rsp(0xffff88025b1d7e48):  0xffffffff8168bbf0  0x0000000043000053  0xffff88026804e480  0x0000000000000003
guest_rsp(0xffff88025b1d7e68):  0x00007f7ab6ffcd38  0x00007f7ab6ffcd38  0xffffffff8128b4f4  0xffffffff82aae7a8
guest_rsp(0xffff88025b1d7e88):  0xffff88025b1d7ed0  0x0000000043000053  0xffff88026804e480  0x0000000043000053
guest_rsp(0xffff88025b1d7ea8):  0x00007f7ab6ffcd38  0x00007f7ab6ffcd38  0xffffffff81442f67  0xffffffff8144da00
guest_rsp(0xffff88025b1d7ec8):  0x0000000000000000  0xffff88026804e481  0xffff88026804e480  0x0000000000000003
guest_rsp(0xffff88025b1d7ee8):  0x0000000043000053  0x00007f7ab6ffcd38  0xffffffff8128bd13  0xffff88025b1d7f58
guest_rsp(0xffff88025b1d7f08):  0x0000000000000010  0xffff88025b214000  0x0000000000000000  0x0000000000000000
guest_rsp(0xffff88025b1d7f28):  0x0000000000000000  0xffffffff810023ce  0x0000000000000000  0x0000000000000000
guest_rsp(0xffff88025b1d7f48):  0x0000000000000000  0xffffffff82000086  0x0000000000000000  0x00000000ffffffff
guest_rsp(0xffff88025b1d7f68):  0x00007f7ab6ffccbc  0x00007f7ab6ffd630  0x00007f7ab6ffcd30  0x000000000000000c
guest_rsp(0xffff88025b1d7f88):  0x0000000000000246  0x000000000044d91f  0x0000000000000024  0x000000000044da2b

Guest Call Trace: **************************************
Maybe the call trace is not accurate, pls check stack!!
[57884704us][cpu=0][sev=3][seq=57]:error[-14] in GVA2GPA, err_code=0x0

mwang106, 2018-11-20:
[External_System_ID] 252864

Mingyuan18, 2018-12-20:
The bug can not reproduced with Hypervisor: acrn-2018w47.4-140000p.

closed 2018-12-20 by Mingyuan18