projectacrn/acrn-hypervisor #1878: The "Using Ubuntu as the Service OS" tutorial is outdated and needs to be refreshed

gvancuts, 2018-11-22:
There are a number of corrections needed in the [Using Ubuntu as the Service OS](https://projectacrn.github.io/latest/tutorials/using_ubuntu_as_sos.html) tutorial. More specifically:
* Service OS kernel should be updated from `linux-pk414-sos-<kernel-version>-<build-version>.x86_64.rpm` to `linux-iot-lts2018-sos-<kernel-version>-<build-version>.x86_64.rpm`
* Refresh the link to use a newer image: https://download.clearlinux.org/releases/24040/clear/x86_64/os/linux-pk414-sos-4.14.57-69.x86_64.rpm
* Add a note that these folders are regularly "purged" so the images are no longer available after a while (so it does not catch the user by surprise)
* Fix the discrepancy between the  SOS kernel installation instructions and the content of the `/etc/grub.d/40_custom` as they indicate different folders
* Refresh the image used as a reference for the UOS
* Update the UOS kernel from `linux-pk414-standard-<kernel-version>-<build-version>.x86_64.rpm` to `linux-iot-lts2018-<kernel-version>-<build-version>.x86_64.rpm`
* Validate if the "Network Sharing" instructions are still valid or if the `systemd` network scripts are installed (and functional) when building and installing `acrn-hypervisor`

gvancuts, 2018-11-23:
@ttzeng , want to take a stab at that?

NanlinXie, 2018-12-04:
Document updated in "https://projectacrn.github.io/latest/tutorials/using_ubuntu_as_sos.html", this issue could be closed.

closed 2018-12-04 by NanlinXie