projectacrn/acrn-hypervisor #1879: dm: mei: ref count for host client is prone to race condition

tomasbw, 2018-11-22:
[Switching to thread 5166 (LWP 11002)]
#0  0x00007f4f5f512c17 in close () from /usr/lib64/libpthread.so.0
(gdb) bt
#0  0x00007f4f5f512c17 in close () from /usr/lib64/libpthread.so.0
#1  0x000056443e1bcd55 in vmei_host_client_destroy (ref=0x56443f489880) at hw/pci/virtio/virtio_mei.c:367
#2  0x000056443e1be337 in refcnt_put (ref=<optimized out>) at
hw/pci/virtio/virtio_mei.c:153
#3  refcnt_put (ref=<optimized out>) at hw/pci/virtio/virtio_mei.c:150
#4  vmei_host_client_put (hclient=<optimized out>) at
hw/pci/virtio/virtio_mei.c:442
#5  vmei_me_client_destroy_host_clients
(mclient=mclient@entry=0x56443f489780) at hw/pci/virtio/virtio_mei.c:473
#6  0x000056443e1be3f8 in vmei_virtual_fw_reset (vmei=0x56443f4cd1e0) at
hw/pci/virtio/virtio_mei.c:917
#7  0x000056443e1be458 in vmei_reset (vth=0x56443f4cd1e0) at
hw/pci/virtio/virtio_mei.c:928
#8  0x000056443e19185a in virtio_pci_legacy_write (ctx=<optimized out>, vcpu=<optimized out>, value=0, size=1, offset=18, baridx=<optimized 
out>, dev=0x56443f4ccf60)
     at hw/pci/virtio/virtio.c:879
#9  virtio_pci_write (ctx=<optimized out>, vcpu=<optimized out>, dev=0x56443f4ccf60, baridx=<optimized out>, offset=18, size=1, value=0) at hw/pci/virtio/virtio.c:1736
#10 0x000056443e1ab4da in pci_emul_io_handler (ctx=<optimized out>, vcpu=<optimized out>, in=<optimized out>, port=<optimized out>, bytes=<optimized out>,
     eax=0x56443e47e258 <vhm_request_page+600>, arg=0x56443f4ccf60) at
hw/pci/core.c:376
#11 0x000056443e1c6e9f in emulate_inout (ctx=<optimized out>, pvcpu=<optimized out>, pio_request=<optimized out>) at core/inout.c:123
#12 0x000056443e1c94e5 in vmexit_inout (ctx=<optimized out>,
vhm_req=0x56443e47e200 <vhm_request_page+512>, pvcpu=<optimized out>) at
core/main.c:304
#13 0x000056443e1c97bc in handle_vmexit (vcpu=<optimized out>,
vhm_req=0x56443e47e200 <vhm_request_page+512>, ctx=0x56443f4a67a0) at
core/main.c:394
#14 vm_loop (ctx=0x56443f4a67a0) at core/main.c:661
#15 start_thread (param=<optimized out>) at core/main.c:246
#16 0x00007f4f5f509108 in ?? () from /usr/lib64/libpthread.so.0
#17 0x00007f4f5f1453cf in clone () from /usr/lib64/libc.so.6
(gdb) thread 1
[Switching to thread 1 (LWP 467)]
#0  0x00007f4f5f06c7cb in raise () from /usr/lib64/libc.so.6
(gdb) bt
#0  0x00007f4f5f06c7cb in raise () from /usr/lib64/libc.so.6
#1  0x00007f4f5f04d535 in abort () from /usr/lib64/libc.so.6
#2  0x00007f4f5f0b4846 in ?? () from /usr/lib64/libc.so.6
#3  0x00007f4f5f0bb64a in ?? () from /usr/lib64/libc.so.6
#4  0x00007f4f5f0bd7cc in ?? () from /usr/lib64/libc.so.6
#5  0x000056443e1bcd6c in vmei_host_client_destroy (ref=0x56443f4cd8b0) at hw/pci/virtio/virtio_mei.c:369
#6  0x000056443e1c698b in mevent_handle (numev=<optimized out>, kev=<optimized out>) at core/mevent.c:174
#7  mevent_dispatch () at core/mevent.c:368
#8  0x000056443e1ca1ef in dm_run (argc=<optimized out>, argv=<optimized 
out>) at core/main.c:969
#9  0x000056443e18cf88 in main (argc=56, argv=0x7ffc3017cdf8) at
core/main.c:1022

wenlingz, 2018-11-23:
@tomasbw please add External_System_ID,thanks

mwang106, 2018-12-24:
Duplicate to  #1875

closed 2018-12-24 by mwang106