projectacrn/acrn-hypervisor #1917: Documentation: What does ',b' flag do with virtio-blk?

mattcurfman, 2018-11-27:
In the launch script @ https://github.com/projectacrn/acrn-hypervisor/blob/master/devicemodel/samples/apl-mrb/launch_uos.sh#L118, there is a ‘,b’ flag added to the virtio-blk parameters for acrn-dm, but this flag is missing in the NUC launch script @ https://github.com/projectacrn/acrn-hypervisor/blob/master/devicemodel/samples/nuc/launch_uos.sh

Further, there is no documentation for this flag at https://projectacrn.github.io/latest/developer-guides/hld/virtio-blk.html

What does the ‘,b’ flag do?

yifanddd, 2019-04-12:
[External_System_ID] ACRN-3378

Mingyuan18, 2019-05-05:
@mattcurfman , Could you please confirm whether this problem still exists? If it does not exist, can you close it?

gvancuts, 2019-05-06:
This issue is still valid as far as I can see, the script uses the `,b` flag as part of the `virtio-blk` options but it is not described anywhere (that I could easily find) in our documentation.

Mingyuan18, 2019-05-24:
Patch merged. So close this issue.

closed 2019-05-24 by Mingyuan18