projectacrn/acrn-hypervisor #1924: DM: Virtual TPM support

YadongQi, 2018-11-28:
1. TPM emulator module
    This module will establish communication channels(unixio socket) with
    swtpm process. It will forward UOS's TPM requests to SWTPM and get
    responses.
    (SWTPM is a libtpms based TPM emulator with socket interface. It is
    opensourced @https://github.com/stefanberger/swtpm. Libtpms is a
    library that targets the integration of TPM functionality into
    hypervisors. Is it opensourced @https://github.com/stefanberger/libtpms.)
2. Virtual TPM CRB device module
    This module mainly monitored the TPM MMIO region(CTRL register and Data
    buffer). It will handle TPM MMIO access from UOS. It will forward command
    to TPM emulator for further processing if UOS want to start a TPM command.
3. TPM wrapper module
    This module wrappers TPM emulator and Virtual TPM CRB device initialization
    functions. It will parse tpm options from launch scripts. If valid argument
    parsed, then it will initialize TPM emulator and virtual TPM CRB device.
4. Build TPM ACPI table for Guest
    If Virtual TPM CRB device is successfully initialized, the ACPI table will
    build proper ACPI tables for Guest.

Mingyuan18, 2018-11-28:
[External_System_ID] ACRN-1941

YadongQi, 2018-11-29:
patches merged, close this issue

closed 2018-11-29 by YadongQi