projectacrn/acrn-hypervisor #1932: SOS resume failed and display black screen when assigning HDMI to SOS and edp to AaaG。

dandanyx, 2018-11-29:
Environment:
------------
SOS version: 26510
HW: MRB

Bug detailed description:
SOS resume failed and display black screen when test S3 suspend and resume

Reproduce steps:
-------------
1. boot up system 
2. boot up Android guest
5. Press the ignition button for 2s, then release button and wait for 20s in order to suspend SOS and UOS.
6. Connect SOS IP by ssh, expected result: failed to connect SOS IP
7. Detect Android guest by adb devices,expected result: failed to detect devices
8. Press the ignition button for 2s, then release button and wait for 20s in order to resume SOS and UOS.

Current result:
--------------
SOS resume failed and display black screen when test S3 suspend and resume

Expected result:
--------------
SOS and UOS can be resumed 
Can connect SOS IP by ssh 
Detect Android guest by adb devices

dandanyx, 2018-11-29:
[External_System_ID] ACRN-1939

Mingyuan18, 2019-01-02:
Can't reproduce this issue in below build: 
ACRN: 2018w51.2.140000p_134
SOS : 26830

closed 2019-01-02 by Mingyuan18