projectacrn/acrn-hypervisor #1957: Race-condition in ioreq_client

yakuizhao, 2018-12-03:
Now the ioreq_client is managed by using static array. It is possible that it is released while it is still accessed in another thread. Then it will cause the unexpected kernel panic.

[34908.005247] BUG: unable to handle kernel NULL pointer dereference at 0000000000000008
[34908.005507] PGD 0 P4D 0 
[34908.005599] Oops: 0000 [#1] PREEMPT SMP
[34908.005732] CPU: 0 PID: 10151 Comm: vcpu 0 Tainted: G     U  W         4.19.0-19.iot-lts2018-sos #1
[34908.006034] RIP: 0010:__list_add_valid+0x0/0x70
[34908.006188] Code: 9a 48 c7 c7 c6 72 49 9a e9 0d ff ff ff 66 66 2e 0f 1f 84 00 00 00 00 00 66 90 48 89 fe 48 c7 c7 d7 72 49 9a e9 f1 fe ff ff 90 <48> 8b 42 08 49 89 d0 48 39 f0 75 18 48 8b 10 4c 39 c2 75 27 48 39
[34908.006799] RSP: 0018:ffff8e74c1d9fbc8 EFLAGS: 00010046
[34908.006976] RAX: ffff8e74c1d9fc38 RBX: 0000000000000001 RCX: ffff8e7732067bd8
[34908.007213] RDX: 0000000000000000 RSI: ffff8e7732067bd8 RDI: ffff8e74c1d9fc38
[34908.007451] RBP: ffff8e74c1d9fc20 R08: 0000000000000000 R09: 000000000000000f
[34908.007688] R10: ffffffff994a2270 R11: ffffffff9ae5bf0d R12: ffff8e7732067bc0
[34908.007925] R13: ffff8e74c1d9fc38 R14: 0000000000000282 R15: 0000000000000001
[34908.008164] FS:  00007fb9013da700(0000) GS:ffff8e7733a00000(0000) knlGS:0000000000000000
[34908.008432] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[34908.008625] CR2: 0000000000000008 CR3: 0000000263ef3000 CR4: 00000000003406f0
[34908.008862] Call Trace:
[34908.008951]  prepare_to_wait_event+0x14d/0x180
[34908.009105]  acrn_ioreq_attach_client+0x228/0x290
[34908.009266]  ? finish_wait+0x80/0x80
[34908.009390]  vhm_dev_ioctl+0x5e4/0xcf0
[34908.009521]  ? _raw_spin_unlock_irqrestore+0x2c/0x60
[34908.009690]  ? ep_poll_callback+0x232/0x2e0
[34908.009835]  ? current_time+0x4f/0x90
[34908.009962]  ? file_update_time+0x5e/0x130
[34908.010104]  ? __sb_end_write+0x45/0x70
[34908.010237]  ? selinux_file_ioctl+0x164/0x200
[34908.010387]  do_vfs_ioctl+0xa5/0x6c0
[34908.010512]  ksys_ioctl+0x60/0x90
[34908.010629]  __x64_sys_ioctl+0x16/0x20
[34908.010759]  do_syscall_64+0x56/0x110
[34908.010886]  entry_SYSCALL_64_after_hwframe+0x49/0xbe
[34908.011058] RIP: 0033:0x7fbd754868e7
[34908.011181] Code: 00 00 90 48 8b 05 99 65 0d 00 64 c7 00 26 00 00 00 48 c7 c0 ff ff ff ff c3 66 2e 0f 1f 84 00 00 00 00 00 b8 10 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 69 65 0d 00 f7 d8 64 89 01 48
[34908.011792] RSP: 002b:00007fb9013d9e38 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[34908.012045] RAX: ffffffffffffffda RBX: 000055a9ea6ef870 RCX: 00007fbd754868e7
[34908.012282] RDX: 0000000000000003 RSI: 0000000043000033 RDI: 0000000000000003
[34908.012520] RBP: 000055a9e8ee2400 R08: 0000000000000000 R09: 000055a9e8ee2058
[34908.012757] R10: 0000000000000000 R11: 0000000000000246 R12: 0000000000000000
[34908.012994] R13: 000055a9e8ee2388 R14: 000055a9e8c50610 R15: 0000000000000004
[34908.013233] Modules linked in: ahci libahci intel_telemetry_debugfs intel_telemetry_pltdrv intel_telemetry_core igb xhci_pci coretemp ici_isys_mod xhci_hcd libata iTCO_wdt mei_me mei ipu4_acpi intel_ipu4_isys_csslib intel_ipu4_psys intel_ipu4_psys_csslib intel_ipu4_mmu intel_ipu4 iova crlmodule_lite
[34908.014115] CR2: 0000000000000008
[34908.014232] ---[ end trace 0f8a5b0ddd861347 ]---

yakuizhao, 2018-12-03:
[External_System_ID] ACRN-1916

anthonyzxu, 2018-12-03:
It is a bug not race-condition.
HVM needs to make sure no one is using it.

Anthony

yakuizhao, 2018-12-04:
Yes. This is a bug. 
Now the user of ioreq_client can be user-space multi-thread or kernel thread. It is difficult to assure that it is not used when it needs to be destroyed(Maybe it is sleeping on the waiting notify). And it increases the complexity of thread synchronization.

So we need to find one mechanism that can assure that the ioreq_client is released correctly while it simplifies the logic of thread synchronization.

yakuizhao, 2018-12-14:
This fix is already merged. So this issue will be closed.

closed 2018-12-14 by yakuizhao